Klipper3d/klipper #2017: cant connect fysetc cheetah 1.2a

saurabhceaseless, 2019-09-28:
Hello

I have been using marlin 2.0 on fysetc cheetah 1.2a and now trying to move over to klipper

However Im facing issue connecting to the mcu after flashing the firmware to the board.

I have followed the installation instructions for octopi for klipper config on a raspberry pi zero w

Here is an overview of my setup -
octopi running on a raspberry pi zero w connecting fysetc cheetah 1.2a via a usb through a CH340/1 usb to serial driver IC which as per documentation works at maz 115200 baud rate
The CH340/1 chip connects over th STM32F103 mcu over USART1 as per fysetc cheetah 1.2a documentation on their github
Using generic fysetc cheetah 1.2 config file renamed as printer.cfg, placed in /home/pi dir with an additional entry under mcu -

[mcu]
baud: 115200
serial: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

These are the options I have choosen for the menuconfig for the firmware compilation -

                        [*] Enable extra low-level configuration options      (had to select baud rate 115200, limited by CH340/1)                                
                        Micro-controller Architecture (STMicroelectronics STM32F1/F4)  --->                 
                        Processor model (STM32F103)  --->                                                
                        Bootloader offset (No bootloader)  --->                                          
                        Clock Reference (8Mhz crystal)  --->                                              
                   [ ] Use USB for communication (instead of serial)   (as specified in klipper generic config file for fysetc cheetah 1.2)                                   
                       Serial Port (USART1)  --->                                                        
                   (115200) Baud rate for serial port       (baud rate changed to 115200 from default 250000)                                              
                   [ ] Specify a custom step pulse duration                                             
                   ()  GPIO pins to set at micro-controller startup 

After make, I flashed the firmware file using the stm32flasher and it shows success (I know its working since I have used the same route to flash the board with marlin 2.0 to rollback).

When starting klipper service, and checking the klippy.log, I get the below output -

Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.7.0-868-g9fcd3e75'
CPU: 1 core ARMv6-compatible processor rev 7 (v6l)
Python: '2.7.13 (default, Sep 26 2018, 18:42:22) \n[GCC 6.3.0 20170516]'
Start printer at Sat Sep 28 05:10:25 2019 (1569643825.8 3872.1)
===== Config file =====
[stepper_x]
step_pin = PB8
dir_pin = !PB9
enable_pin = !PA8
step_distance = .0125
endstop_pin = ^PA1
position_endstop = 0
position_max = 200
homing_speed = 50

[tmc2208 stepper_x]
uart_pin = PA12
tx_pin = PA11
microsteps = 16
run_current = 0.800
hold_current = 0.500
stealthchop_threshold = 250

[stepper_y]
step_pin = PB2
dir_pin = !PB3
enable_pin = !PB1
step_distance = .0125
endstop_pin = ^PB4
position_endstop = 0
position_max = 200
homing_speed = 50

[tmc2208 stepper_y]
[... 145 lines cut ...]
Recv: ok
Changing monitoring state from "Connecting" to "Operational"
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: ok FIRMWARE_VERSION:v0.7.0-868-g9fcd3e75 FIRMWARE_NAME:Klipper
Send: N2 M21*18
Recv: ok
[...]
Send: status
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: // Klipper state: Not ready
Recv: !! Printer is not ready
Recv: ok

I have tried using baud rate 250000 as well but getting the same issue.

Please help as I'm running out of options to try.

I'm able to rollback to marlin for now and its working fine, printer is working as expected.

Thanks
Saurabh

saurabhceaseless, 2019-09-28:
btw, I have checked on the usb connection, the serial port path as specified in config is accurate and tried re-building the driver for CH340/1 as well

saurabhceaseless, 2019-09-28:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3665113/klippy.log)
Attaching klippy.log

saurabhceaseless, 2019-09-28:
Hello

I tried again by making the firmware file with baud rate 25000, no usb(ofcourse) and no bootloader and this time flashed the firmware using octoprint's firmware flasher plugin
It is strange to me as when I was flashing earlier from cmd using stm32flash, I had used the exact command args as mentioned in the generic fystec cheetah 1.2 config file, below is the output from command line -

root@octopi:/home/pi/klipper# stm32flash -w out/klipper.bin -v -i rts,-dtr,dtr /dev/ttyUSB0
stm32flash 0.5

http://stm32flash.sourceforge.net/

Using Parser : Raw BINARY
Interface serial_posix: 57600 8E1
Version      : 0x22
Option 1     : 0x00
Option 2     : 0x00
Device ID    : 0x0414 (STM32F10xxx High-density)
- RAM        : 64KiB  (512b reserved by bootloader)
- Flash      : 512KiB (size first sector: 2x2048)
- Option RAM : 16b
- System RAM : 2KiB
Write to memory
Erasing memory
Wrote and verified address 0x08004218 (100.00%) Done.

I got the serial connection working with the board this way, but now ran in to another problem, the z stepper tmc2208 connection is not working, error says tmc uart not working; attaching klippy.log for this.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3665116/klippy.log)

I have verified the pin config from the fysetc's provided pins.h from marlin 2.0 and whats in the printer.cfg file, everything seems fine here pin wise.

Thanks
Saurabh

saurabhceaseless, 2019-09-28:
Hello

Tried by commenting tx_pin as well in the config file and tried again, same result.

Something seems to be missing in the config. I checked in the extra config file to find if I’m missing anything necessary for configuring tmc2208 but everything seems to be in place.

Thanks
Saurabh

qldestlp, 2019-09-29:
Right now I have the same problem with the SKR DIP board. Fixed a config that was suitable for SKR MINI. But the connection did not reach (

saurabhceaseless, 2019-09-29:
I’m still struck with this, board connects over serial connection but then kliyopetra reports that it can’t connect to tmc drivers over uart 
Problem is can’t diagnose where the problem is, the config file or the way the octopi connects to the mcu

KevinOConnor, 2019-09-29:
It looks to me that the sample config has the Z uart pins backwards - it should be:
```
uart_pin: PB10
tx_pin: PB11
```

If that change is made, does it improve things?  If not, and you comment out the Z tmc2208 section, does the rest of the printer otherwise come up correctly?

-Kevin

saurabhceaseless, 2019-09-30:
It worked perfectly after corrected pins, I suppose that was the only issue.

KevinOConnor, 2019-09-30:
Okay, thanks.  Commit db2ee2c0.

-Kevin

saurabhceaseless, 2019-09-30:
closing this issue, its fixed to satisfaction.

closed 2019-09-30 by saurabhceaseless