Klipper3d/klipper #2019: mcu failure on bed leveling and screw calibration

andreklug, 2019-09-28:
I have started using Klipper recently. Prints were looking good already, I used SCREWS_TILT_CALCULATE and SCREWS_TILT_CALCULATE to level the bed perfectly. 

Then I went on to change the TMC2208drivers from standalone to UART, for which I removed all the Jumpers on my RAMPS 1.4 board (MS1, MS2, MS3 per stepper driver), the UART connection works. After having that set, I started to tune the pressure advance. Somewhere in the process I bumped into the printer which led to the bed being not leveled anymore, at least that's what I think. So I wanted to start the SCREWS_TILT_CALCULATE again.

But now whenever after homing I try one of the two functions, I am getting the following error in klippy.log, and mcu is shut down:

```
stepcompress o=7 i=0 c=597 a=0: Invalid sequence
Internal error on command:"SCREWS_TILT_CALCULATE"
Traceback (most recent call last):
  File "/home/pi/Klipper/klippy/gcode.py", line 224, in _process_commands
    handler(params)
  File "/home/pi/Klipper/klippy/gcode.py", line 74, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/Klipper/klippy/extras/screws_tilt_adjust.py", line 60, in cmd_SCREWS_TILT_CALCULATE
    self.probe_helper.start_probe(params)
  File "/home/pi/Klipper/klippy/extras/probe.py", line 340, in start_probe
    pos = probe.run_probe(params)
  File "/home/pi/Klipper/klippy/extras/probe.py", line 120, in run_probe
    pos = self._probe(speed)
  File "/home/pi/Klipper/klippy/extras/probe.py", line 72, in _probe
    probe_pos=True, verify_movement=verify)
  File "/home/pi/Klipper/klippy/homing.py", line 58, in homing_move
    self.toolhead.drip_move(movepos, speed)
  File "/home/pi/Klipper/klippy/toolhead.py", line 448, in drip_move
    self.move_queue.add_move(smove)
  File "/home/pi/Klipper/klippy/toolhead.py", line 193, in add_move
    self.flush(lazy=True)
  File "/home/pi/Klipper/klippy/toolhead.py", line 181, in flush
    move.move()
  File "/home/pi/Klipper/klippy/toolhead.py", line 110, in move
    self.accel_t + self.cruise_t + self.decel_t)
  File "/home/pi/Klipper/klippy/toolhead.py", line 281, in update_move_time
    m.flush_moves(flush_to_time)
  File "/home/pi/Klipper/klippy/mcu.py", line 802, in flush_moves
    self._name,))
error: Internal error in MCU 'mcu' stepcompress
Internal error on command:"SCREWS_TILT_CALCULATE"
```
Printing still works in general.
I already tried removing the UART setup, deactivating (commenting out, and setting to 0) the pressure advance setting in the printer.cfg, re-flashing the firmware and switching back to a printer.cfg from the time it worked. No success.

Thanks in advance
[klippy.log](https://github.com/KevinOConnor/klipper/files/3665664/klippy.log)

 for your help!

KevinOConnor, 2019-09-29:
The root cause of the error was a "Rescheduled timer in the past" - which results when the micro-controller is asked to process more than it can.  The root cause of that is the probe speed of 30 in your config - Klipper properly slowed the z to 3mm/s, but it scheduled endstop checking as if the Z was moving at 30mm/s - which the atmega2560 was unable to keep up with.

Klipper should have done better here.  In the meantime, reduce your probe speed to match your maximum z speed.

-Kevin

andreklug, 2019-09-30:
Thank you, Kevin.

At some point yesterday I reduced the Z axis speed and was able to get both functions working again. It would be great if the error message would be more verbose in this situation.
However, I am not able to configure the Z Axis right, and am confused when calculating the step_distance.

(btw I am using a Prusa i3 clone)

For X/Y it works:

```
[stepper_x]
step_pin: ar54
dir_pin: ar55
enable_pin: !ar38
step_distance: .0125
endstop_pin: ^!ar3
position_endstop: 0
position_min: -3
position_max: 200
homing_speed: 80
homing_retract_dist: 5

[tmc2208 stepper_x]
uart_pin: ar63
microsteps: 16
interpolate: false
run_current: 0.700
hold_current: 0.355
sense_resistor: 0.110
stealthchop_threshold: 300
#driver_IHOLDDELAY: 8
#driver_TPOWERDOWN: 20
#driver_TBL: 2
#driver_TOFF: 3
#driver_HEND: 0
#driver_HSTRT: 5
#driver_PWM_AUTOGRAD: True
#driver_PWM_AUTOSCALE: True
#driver_PWM_LIM: 12
#driver_PWM_REG: 8
#driver_PWM_FREQ: 1
#driver_PWM_GRAD: 14
#driver_PWM_OFS: 36
```
According to the Marlin config, X has 80 steps per mm. So I just set 1/80 = 0.0125 for the step distance. However, when I take the microsteps = 16 into consideration, I don't see how I would get into this result. But of course they matter, because if I change them to 4 for example, I need another step_distance.

For Z this doesn't work:

```
[stepper_z]
step_pin: ar46
dir_pin: ar48
enable_pin: !ar62
step_distance: .000625
endstop_pin: probe:z_virtual_endstop
position_min: -5
position_max: 180
homing_retract_dist: 5

[tmc2208 stepper_z]
uart_pin: ar42
microsteps: 4
interpolate: false
run_current: 0.700
#run_current: 0.548
hold_current: 0.355
#hold_current: 0.355
sense_resistor: 0.110
stealthchop_threshold: 300
```

Marlin states 4000 steps per mm. 1/4000=0,00025. Setting this value let's the motors scream at 4 microsteps. However, in between I had good results with 0,001. (Maybe because I got to multiply by 4 Microsteps? I don't get the math, when I take the calculation above into consideration). I don't know what's different now, but with this setting printing works, although the print has only 50% height. If I set it to 0.000625 or 0.0005 -> screams (also after setting Z axis speed down to 1 mm/sec and acc to 10mm / sec).

Thanks for your help!
KR

jakep82, 2019-09-30:
With a common 20 tooth pulley, your X and Y axis will move 1mm for every 5 full steps.  At 1/16 microstepping, that is 5*16=80 microsteps per mm.  This is why Marlin uses 80 steps per mm, and the microstepping is already calculated in that number.

If you're using a Prusa clone, you likely have an 8mm pitch leadscrew on your Z axis.  That means 200 full steps, or 3200 microsteps at 1/16 moves the axis 8mm.  3200/8=400 microsteps per mm.  That means your step_distance should be 1/400=.0025 for 1/16 microstepping.

https://blog.prusaprinters.org/calculator/

andreklug, 2019-10-01:
Thank you @jakep82 !

closed 2019-10-01 by andreklug