Klipper3d/klipper #2022: LCD character navigation panel not working with Klipper and Repetier-server

dragsterbox, 2019-09-29:
Hi everyone,

I finished the installation of repetier-server with Klipper. Almost everything is working fine: printer is connected on repetier server, I can control printer from repetier's server page and klipper seems to be responding.

Regarding the LCD character panel, information are displayed correctly and refreshing when for e.g. I turn on the heat bed but I cannot use the navigation knob and navigate to the menu.

When I install Octoprint, I have no issue at all with the LCD.

Some help will be very appreciated.

[image]

[klippy.log](https://github.com/KevinOConnor/klipper/files/3666418/klippy.log)

# This file contains a configuration for the "Anycubic Kossel Linear
# Plus Large Printing Size", "Anycubic Kossel Pulley Plus Large
# Printing Size" and similar delta printer from 2017.
# The Anycubic delta printers use the TriGorilla board which is an
# AVR ATmega2560 Arduino + RAMPS compatible board.
# To use this config, the firmware should be compiled for the AVR atmega2560.

# See the example.cfg file for a description of available parameters.

[stepper_a]
step_pin: ar54
dir_pin: ar55
enable_pin: multi_pin:my_enable_pin
step_distance: .010000
endstop_pin: ^ar2
homing_speed: 50
# The next parameter needs to be adjusted for
# your printer. You may want to start with 280
# and meassure the distance from nozzle to bed.
# This value then needs to be added.
position_endstop: 273.6
angle: 210.00000
arm_length: 267

[tmc2208 stepper_a]
uart_pin: ar4
microsteps:16
interpolate: True
run_current: 0.800
hold_current: 0.500
sense_resistor: 0.110
stealthchop_threshold: 999

[stepper_b]
step_pin: ar60
dir_pin: ar61
enable_pin: multi_pin:my_enable_pin
step_distance: .010000
endstop_pin: ^ar15
position_endstop: 273.6
angle: 330.000000
arm_length: 267

[tmc2208 stepper_b]
uart_pin: ar5
microsteps:16
interpolate: True
run_current: 0.800
hold_current: 0.500
sense_resistor: 0.110
stealthchop_threshold: 999

[stepper_c]
step_pin: ar46
dir_pin: ar48
enable_pin: multi_pin:my_enable_pin
step_distance: .010000
endstop_pin: ^ar19
position_endstop: 273.6
angle: 90.000000
arm_length: 267

[tmc2208 stepper_c]
uart_pin: ar6
microsteps:16
interpolate: True
run_current: 0.800
hold_current: 0.500
sense_resistor: 0.110
stealthchop_threshold: 999

[extruder]
step_pin: ar26
dir_pin: !ar28
enable_pin: multi_pin:my_enable_pin
#ORGINAL step_distance: 0.010989
step_distance: .004927
nozzle_diameter: 0.400
filament_diameter: 1.750
max_extrude_only_distance: 100.0
heater_pin: ar10
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog13
control: pid
pid_Kp: 25.349
pid_Ki: 1.216
pid_Kd: 132.130
min_extrude_temp: 150
min_temp: 10
max_temp: 290
max_extrude_only_accel: 10000

max_extrude_only_distance: 650.0

pressure_advance: 0.25

pressure_advance_lookahead_time: 0.010000
#  do not use pressure advance until you read about it and tune for it
#  this is for 100mm/s or higher printer settings.

[tmc2208 extruder]

uart_pin: ar11
microsteps:8
run_current:0.750
hold_current:0.500
stealthchop_threshold: 999

[heater_bed]
heater_pin: ar8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog14
control: watermark
min_temp: 10
max_temp: 130

[fan]
pin: ar9
#kick_start_time: 0.200

[heater_fan extruder_cooler_fan]
pin: ar44
heater: extruder
heater_temp: 50.0
fan_speed: 1.0

[multi_pin my_enable_pin]
pins: !ar38, !ar56, !ar62, !ar24, ar7
#Starts the stepper cooling fan when the A stepper is activated

[mcu]
serial: /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0
baud: 250000
pin_map: arduino

[printer]
kinematics: delta
max_velocity: 400
max_accel: 3000
max_z_velocity: 100
minimum_z_position: -5.0
delta_radius: 115

[idle_timeout]
timeout: 360

[probe]
pin: ^!ar18
# Probe detection pin. This parameter must be provided.
x_offset: 0.0
#   The distance (in mm) between the probe and the nozzle along the
#   x-axis. The default is 0.
y_offset: 0.0
#   The distance (in mm) between the probe and the nozzle along the
#   y-axis. The default is 0.speed: 10.0
z_offset: -00.1200
#   The Z height (in mm) of the head when the probe triggers.
speed: 12
# The distance (in mm) to retract between each sample if sampling
# more than once. The default is 2mm.
samples: 3
# be averaged. The default is to probe 1 time.
sample_retract_dist: 5

[gcode_macro set_delta_height]

gcode:
	M117 SET_DELTA_HEIGHT	
	G28
	G0 F4000 X0 YO Z10 
	PROBE
	G0 F2000 Z3
	PROBE
	G92 Z-00.1200
	G0 F4000 Z10

[gcode_macro endstop_phase]

gcode:
    G28
    ENDSTOP_PHASE_CALIBRATE
    G1 Z200 F2000
    G28
    G1 Z180 Y70 F2000
    G28
    G1 Z220 X70 F2000
    G28
    G1 Z210 Y-55 F2000
    G28
    G1 Z150 X-45 F2000
    G28
    G1 Z200 X-20 Y-45 F2000
    G28
    ENDSTOP_PHASE_CALIBRATE STEPPER=stepper_a
    ENDSTOP_PHASE_CALIBRATE STEPPER=stepper_b
    ENDSTOP_PHASE_CALIBRATE STEPPER=stepper_c
    SAVE_CONFIG

[delta_calibrate]
radius: 115
# manual_probe: false
# If true, then DELTA_CALIBRATE will perform manual probing. If
# false, then a PROBE command will be run at each probe
# point. Manual probing is accomplished by manually jogging the Z
# position of the print head at each probe point and then issuing a
# NEXT extended g-code command to record the position at that
# point. The default is false if a [probe] config section is present
# and true otherwise.
speed: 100
# The speed (in mm/s) of non-probing moves during the calibration.
# The default is 50.
horizontal_move_z: 25
# The height (in mm) that the head should be commanded to move to
# just prior to starting a probe operation. The default is 5.
# The number of times to probe each point. The probed z-values will

# "RepRapDiscount 2004 Smart Controller" type displays
[display]
lcd_type: hd44780
rs_pin: ar16
e_pin: ar17
d4_pin: ar23
d5_pin: ar25
d6_pin: ar27
d7_pin: ar29
encoder_pins: ^ar31, ^ar33
click_pin: ^!ar35
kill_pin: ^!ar41

# A virtual sdcard may be useful if the host machine is not fast
# enough to run OctoPrint well. It allows the Klipper host software to
# directly print gcode files stored in a directory on the host using
# standard sdcard G-Code commands (eg, M24).
[virtual_sdcard]
path: ~/.octoprint/uploads/
#   The path of the local directory on the host machine to look for
#   g-code files. This is a read-only directory (sdcard file writes
#   are not supported). One may point this to OctoPrint's upload
#   directory (generally ~/.octoprint/uploads/ ). This parameter must
#   be provided.

# Stepper phase adjusted endstops. To use this feature, define a
# config section with an "endstop_phase" prefix followed by the name
# of the corresponding stepper config section. This feature can 
# improve the accuracy of endstop switches. Add a bare 
# "[endstop_phase]" declaration to
# enable the ENDSTOP_PHASE_CALIBRATE command.
[endstop_phase]
#   This specifies the number of phases of the given stepper motor
#   driver (which is the number of micro-steps multiplied by four).
#   This setting is automatically determined if one uses TMC2130,
#   TMC2208, TMC2224 or TMC2660 drivers with run-time configuration.
#   Otherwise, this parameter must be provided.

####START CODE SLIC3R PE
#M117 HERE WE GO..
#SET_PIN PIN=led VALUE=255
#M140 S[first_layer_bed_temperature] ; set bed temp
#G28
#M104 S150 ;PRE HEAT HOTEND
#M190 S[first_layer_bed_temperature] ; wait for bed temp
#M109 S150
#set_delta_height
#M117 READY..
#M104 S[first_layer_temperature]
#G1 X-74 Y-43 Z50 F1000 ; move  nozzle to initial point
#M117 SET..
#M109 S[first_layer_temperature];  wait for TEMP to be reached
#M117 GO.....
#G92 E0 ; zero the extruded length
#G1 X-44 Y-75 Z0.2 E1.5 F3000 ; Drop to bed
#M117 SQUIRTING...
#G1 X0 Y-90 E10 F500 ; Extrude 14mm of filament in a 4cm line
#G1 X44 Y-75 Z0.3 E12 F500 ; Extrude 12mm of filament in a 4cm line
#G92 E0 ; zero the extruded length
#G1 E-0.4 F500 ; Retract a little
#G1 X87 Y-17 Z1 F6000 ; Quickly wipe away from the filament line
#G1 Z5 ; Raise and begin printing.
#M117 DOODELING...

###START OF LAYER
# M117 LAYER # [layer_num]  @ [layer_z]mm

##END CODE SLIC3R PE
# M117 TADA!!!...
# G1 E-4 F500;retract filament
# M104 S0 ; turn off temperature
# M140 S0 ; turn off heatbed
# G91;Relative Positioning 
# G1 Z10 F8000
# G90 ;Absolute Positioning 
# G1 X0 Y0 F8000 ;GO TO CENTER
# G28
# M84     ; disable motors
# M117 All done. BYE...
# SET_PIN PIN=led VALUE=0

# Delta max build height 275.62mm (radius tapered above 251.93mm)
# Delta max build radius 99.58mm (moves slowed past 108.94mm and 117.23mm)
# Extruder max_extrude_ratio=0.266081

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [endstop_phase stepper_a]
#*# endstop_phase = 33
#*#
#*# [endstop_phase stepper_b]
#*# endstop_phase = 25
#*#
#*# [endstop_phase stepper_c]
#*# endstop_phase = 35

KevinOConnor, 2019-09-29:
Your log file indicates Klipper failed to start up correctly (the config points to a virtual sdcard directory that doesn't exist).  The display will only report status after Klipper fully starts up correctly.  Be sure to correct any errors in the config before trying to use the display.

-Kevin

dragsterbox, 2019-09-30:
Thanks a lot Kevin. If fixed the issue and created the folder "/home/pi/.octoprint/uploads" to make Klipper work.

closed 2019-10-08 by KevinOConnor