Klipper3d/klipper #2024: Option to discard first read value when using multiple probes

vladbabii, 2019-09-29:
## Short version

I would like to request an option, when probing or homing (so for all XYZ endstops) to configure klipper to discard first value of the homing/probing. Based on my tests, if the kinematics of the printer are good enough, if something varies it's most likely in the first value read. By ignoring / discarding the first value a much better reading can be had. 

For example, with tmc2208 in standalone mode, at 50+ mm/s only the first hit of the endstop varied with +1 +0 micro step. If the first value can be discarded then the precision would be much better up to higher speeds.

## Long version:

I built a jig to test endstop precision
- steppers: a4988, drv8825, tmc2208, tmc2209 (tmcs both uart and without)
- endstops: omron high precision, omron normal precision, various no name (chinese) mechanical switches, optical endstops
[image]

Ran the PROBE_ACCURACY command to test things.

Take-aways:
- optical endstops are worse than almost any option at over 30 mm/s (2+ steps can be missed)
- omron switches work with same precision (direct or with metal blade over button) up to 40 mm/s
- high precision switches work ok up to 70 mm/s
- when doing multiple probes, the error usually appears only in first probe so if possible, discard first reading for much higher accuracy
- most mechanical switches had 1 step error rate between 30 and 50 mm/s

[klippy.zip](https://github.com/KevinOConnor/klipper/files/3668150/klippy.zip)

KevinOConnor, 2019-10-03:
FYI, homing normally does a "double tap" of the endstop - it's configured via the `homing_retract_dist` and `second_homing_speed` parameters.

For probing, it's possible to use the `samples_tolerance` and `samples_tolerance_retries` options to retry on a bad probe.

-Kevin

vladbabii, 2019-10-03:
But don't sample_tolerance and samples_tolerance_retries retry second/third measurement ? I want to discard first one.

KevinOConnor, 2019-10-03:
If the `samples: 2` then it should be possible to detect the discrepancy on the second probe, and then a total of 4 probes would be made (the first two discarded).  Dropping the first probe attempt outright might be possible from g-code scripts - otherwise a developer would need to implement and submit the code to implement it.

-Kevin

KevinOConnor, 2019-10-19:
I'm going to close this as it looks like the conversation has concluded.  The existing config options can provide similar functionality and/or a developer can extend the code.

-Kevin

closed 2019-10-19 by KevinOConnor