Klipper3d/klipper #2025: MCU 'mcu' shutdown: Timer too close

3lementalS, 2019-09-30:
I was working on a print to happen overnight, woke up a few hours in a saw that the print stopped and gave me this error, this is not the first time I got this same error. Hoping I have the full log.  I followed the instructions but could not find the exact directory.

[klippy.zip](https://github.com/KevinOConnor/klipper/files/3668409/klippy.zip)

KevinOConnor, 2019-10-03:
That is the correct log.  The log indicates the host software was unable to get cpu time for an extended period (300+ milliseconds).  I've not seen this before.

Is there anything else running on the host machine that could be causing high load or any custom software installed that could be changing system priorities?

-Kevin

3lementalS, 2019-10-03:
I did have some plugins like themify installed.

KevinOConnor, 2020-01-23:
Okay, I'm going to close this as it does not appear to be a widespread issue.  I'm unsure why the issue occurred in the original report.  If it continues to occur, please provide the log and I will reopen the investigation.

-Kevin

closed 2020-01-23 by KevinOConnor