Klipper3d/klipper #2026: fysetc cheetah does not reconnects after restarting klipper service

saurabhceaseless, 2019-09-30:
The Klipper fails to re-connect to fysetc cheetah board after restarting klipper service post making any config change.

Either the error stating 'identity response error' or gives Unable to open port  [Errno 2] could not open port /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0'

I have a couple of board configured together, skr 1.3 and a fysetc cheetah 1.2
Skr1.3 is able to connect after service restart but fysetc cheetah fails to do so.
Bad thing fysetc cheetah does not have a reset button on board and reset pin on mcu is not exposed anywhere on the pcb. My only option left after this is to power down and restart the entire printer.

Im sharing my config file and klippy.log for your reference.

Thanks
Saurabh
[klippy.log](https://github.com/KevinOConnor/klipper/files/3670926/klippy.log)

BlackStump, 2019-09-30:
have you looked here?
https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L274
 and tried adding under 
[mcu]
restart_method: command

saurabhceaseless, 2019-09-30:
Yes, tried, I found out toggling thru these settings that by default klipper uses command method to restart the mcu if it cannot find a serial connection
You can see in the klippy log attached, it actually does sends a restart to both the boards yet cant find connection back to either
Whether board actually does gets restarted at this point or not? Not certain
But if it does, then klipper is restarting probably too fast for the board to come up back again and acquire a successful USB connection to raspberry pi.
If it doesnt, then other options left is arduino and usb
arduino <- cant do since none of my boards is an arduino 
usb <- only skr 1.3 is dependent(can be made) on the usb power, fysetc cheetah 1.2 mcu is powered internally so this wont work with fysetc cheetah

Thanks
Saurabh

saurabhceaseless, 2019-09-30:
I cant really understand the meaning of below log output, where it first gets the response from the mcu and then it attempts to reset both -

Starting serial connect
got {'#receive_time': 589.770925, u'next_clock': 4180974336L, u'oid': 4, u'value': 30983, '#name': u'analog_in_state', '#sent_time': 589.7637629999999}
Loaded MCU 'mcu' 77 commands (v0.7.0-868-g9fcd3e75-20190928_055223-octopi / gcc: (15:5.4.1+svn241155-1) 5.4.1 20160919 binutils: (2.27-9+9) 2.27)
MCU 'mcu' config: BUS_PINS_i2c1=PB6,PB7 RECEIVE_WINDOW=192 BUS_PINS_i2c2=PB10,PB11 BUS_PINS_spi2=PB14,PB15,PB13 SERIAL_BAUD=250000 STEP_DELAY=2 ADC_MAX=4095 CLOCK_FREQ=72000000 BUS_PINS_spi1=PA6,PA7,PA5 MCU=stm32f103xb STATS_SUMSQ_BASE=256 RESERVE_PINS_serial=PA10,PA9
Configured MCU 'mcu' (1024 moves)
Starting serial connect
Loaded MCU 'bt' 77 commands (v0.7.0-868-g9fcd3e75-20190930_080532-octopi / gcc: (15:5.4.1+svn241155-1) 5.4.1 20160919 binutils: (2.27-9+9) 2.27)
MCU 'bt' config: BUS_PINS_i2c1=P0.1,P0.0 STEP_DELAY=2 ADC_MAX=4095 BUS_PINS_ssp0=P0.17,P0.18,P0.15 RESERVE_PINS_USB=P0.30,P0.29,P2.9 CLOCK_FREQ=100000000 MCU=lpc1768 STATS_SUMSQ_BASE=256
Attempting automated MCU 'bt' restart: CRC mismatch
Attempting MCU 'mcu' reset via rpi usb power
Attempting MCU 'bt' reset command
Restarting printer

Thanks
Saurabh

KevinOConnor, 2019-09-30:
Set `restart_method: command` for the fysetc and do not set restart_method for the skr.  If that doesn't work, attach the full Klipper log file from the attempt.

-Kevin

saurabhceaseless, 2019-09-30:
Tried above, now it first throws a protocol error, then says cant exclusively lock the port, then a timeout and then finally identity response error.

Attaching log and config file.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3671888/klippy.log)

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/3671889/printer.cfg.txt)

Thanks
Saurabh

saurabhceaseless, 2019-09-30:
I do see both the boards connected on the serial though, klipper is unable to connect to it

root@octopi:/tmp# ls /dev/serial/by-id/
usb-1a86_USB2.0-Serial-if00-port0  usb-Klipper_Klipper_firmware_12345-if00

KevinOConnor, 2019-10-01:
It looks like the Cheetah uses the dtr/rts lines to enter the bootloader while the "arduino" reboot mechanism uses the dtr to request a reboot.  The two mechanisms appear to not be compatible and an arduino style request may put the Cheetah in an unstable state.  After making the config update (so that the Cheetah never gets the arduino style request), do a full software restart, followed by a full hardware restart (so as to clear any past incorrect state).  Does that improve things?  If not, can you attach that log?

-Kevin

saurabhceaseless, 2019-10-02:
Okay, so here I have made a fresh start, made a config change first(did it a day before) and then powered on the system the next day(got busy with work) again did a config change(changed z offset for BLtouch) and then restarted the klipper service for new config to update. Getting identity response error. I have attached the logs and my latest config -

[klippy.log](https://github.com/KevinOConnor/klipper/files/3679701/klippy.log)
[printer.cfg_latest.txt](https://github.com/KevinOConnor/klipper/files/3679702/printer.cfg_latest.txt)

Thanks
Saurabh

KevinOConnor, 2019-10-03:
Your log indicates the config file lacked the required `restart_method: command` in the mcu config section.  Add the config setting and then do a full hardware restart.

-Kevin

saurabhceaseless, 2019-10-04:
Hello

Sorry my mistake, I have re-tried after following the above steps, please find attached my latest config file and klippy log.

After sending the restart command and a service restart of host, it just reports that mcu is in shutdown state.

Thanks
Saurabh
[klippy.log](https://github.com/KevinOConnor/klipper/files/3690808/klippy.log)
[printer.cfg_Latest.txt](https://github.com/KevinOConnor/klipper/files/3690809/printer.cfg_Latest.txt)

KevinOConnor, 2019-10-04:
You have a totally separate error in your config now - the enable pin in the extruder section has an error.  You've also got a warning about changes in the file after the save config section.  I suggest you clean up the config and verify everything comes up correctly on a fresh power-up.  Only then verify the RESTART capability.

-Kevin

saurabhceaseless, 2019-10-04:
[klippy (1).log](https://github.com/KevinOConnor/klipper/files/3691776/klippy.1.log)

Sorry I copied the entire log, please check now.

I was making some changes to some pin assignments on the two boards prior to checking for the restart issue which was up in pipeline in my todo list. Somehow this log also crept in to the log I was supposed to sent you.

The log I have attached now concerns only the restart issue. Please check now.

Thanks
Saurabh

KevinOConnor, 2019-10-04:
Your logs continue to show unrelated issues.  Please perform a full power down, then power up.  If you continue to have issues after that, please provide the log from a clean power up.

-Kevin

KevinOConnor, 2019-10-19:
Any further updates on this?

-Kevin

saurabhceaseless, 2019-10-20:
Still on the exact same state.

All I’m doing is restarting the klipper service from octopi ssh console and identity response error shows up in klippy.log

A complete hardware reset by which I mean kill the power to the printer and connect back again fixes it.

Even when I do the mcu restart from the octoprint klipper plugin same problem follows.

I’ll redo the above process to capture the logs and then I’ll share them with you. The issue is 100% reproducible.

Thanks
Saurabh

saurabhceaseless, 2019-10-24:
Hello

Any thing further I can do about this?
Is this confirmed as a bug?

Thanks
Saurabh

saurabhceaseless, 2019-10-24:
Hello

Request you to please describe clean power up, I'll replicate the steps and share the log with you.

Thanks
Saurabh

KevinOConnor, 2019-10-24:
1 - Make sure the Klipper config file is in the requested state (specifically that `restart_method: command` is specified for the cheetah board).
2 - Power down the raspberry pi.
3 - Remove power from all micro-controllers.  Physically disconnect any plugs that could provide power to them.
4 - wait 30 seconds
5 - provide power to the raspberry pi
6 - provide power to the micro-controllers
7 - wait until everything comes up and you get proper status from OctoPrint
8 - issue a `RESTART` command in the OctoPrint terminal window
9 - attach the resulting klipper log file here.

saurabhceaseless, 2019-11-09:
Hello Kevin

I followed the exact steps as stated by you above -

Once I got the Octopi all up and running -
[image]

I issued a restart command from the octopi terminal, terminal output below -
Connecting to: /tmp/printer
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x69bf8ef0, open=True>(port='/tmp/printer', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from "Connecting" to "Operational"
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: ok FIRMWARE_VERSION:v0.7.0-868-g9fcd3e75-dirty FIRMWARE_NAME:Klipper
[...]
Send: RESTART
Recv: ok
Recv: // Klipper state: Disconnect

Soon after the service restarted, I got identity response error in the klippy log, attaching log below -
[klippy.log](https://github.com/KevinOConnor/klipper/files/3826851/klippy.log)

Thanks
Saurabh

KevinOConnor, 2019-11-14:
Very odd that the lpc176x shows a config change during the restart.  However, that aside, I don't know why the board does not come back up.  Someone with the hardware will need to diagnose what's going on that causes the board to lock-up. 

-Kevin

grigorig, 2020-01-31:
I got that board (while it was cheap) and had the very same problem. Fortunately, it's not hard to figure out what's going wrong.
I suspected the logic to trigger resets and bootloader from the very start, and it is indeed the culprit.
DTR is not special - it's basically Arduino-like. However, RTS controls the BOOT0 pin, with positive logic: that is, whenever RTS is set, BOOT0 is set to HIGH. If there is an MCU reset, it will end up in the bootloader. And that's the problem!

Linux normally sets both RTS and DTR when opening a serial port. So an Arduino-like reset (the default) will always end up in the bootloader. The command restart method doesn't work either, because guess what, Klipper needs to open the serial port for that, which sets RTS.

A quick hack that seems to work for me is resetting RTS directly after Klipper opens the serial port and switching to the command restart method.

@KevinOConnor What do you think might be the best way to deal with this?

grigorig, 2020-01-31:
The DTR/RTS situation is actually a bit more complicated. pySerial has a particular behavior of always setting DTR/RTS by default. Looks like Linux by itself doesn't normally fiddle with RTS, at least. I'm investigating it right now.

KevinOConnor, 2020-01-31:
Okay, thanks.  I thought it was possible to set the dtr/rts flags prior to opening the port on pySerial, but I haven't looked recently.  Let us know what you find.

-Kevin

grigorig, 2020-01-31:
Yes, that is possible, if you do not specify any port when creating the `Serial` instance.
Do you know if any other board cares about the RTS signal? If not, maybe it's fine to get away with always resetting RTS on connect.

closed 2020-04-02 by KevinOConnor

csutcliff, 2020-04-06:
FWIW this seems to be necessary on my 1.1b too.

KevinOConnor, 2020-04-08:
Okay, I added it to generic-fysetc-cheetah-v1.1.cfg as well (commit 03fb7961).

-Kevin

commit 6d2f3ef4 (2020-04-02): config: Fix reset/restart on Fysetc Cheetah v1.2

Deassert RTS for this board and enable the new Cheetah-specific reset
sequence.

Fixes #2026.