Klipper3d/klipper #2027: Heater extruder not heating at expect rate

jakevsnake, 2019-09-30:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3672209/klippy.log)

I keep getting random "Heater extruder not heating at expected rate" fails during my prints.  The most recent happened on a gcode that printed successfully prior to this failure.  I thought the thermistor might be bad but the klippy.log shows it working just fine.  Something commanded the target temp to zero and I dont know why.

andreklug, 2019-10-01:
I am having that sometimes when I start a new job short after the heater has turned off. There is a timeout set when the temperature should have raised noticeable, and for me this is to short because the heater just cools down a few more seconds before the heating kicks in. I believe I read somewhere about defining the timeout duration but didn't find it again. Maybe it's the same for you.

jakevsnake, 2019-10-01:
I may have found my issue.  I had a thermistor wire pull out of the back of the connector when I tugged on it.  I changed the thermistor and the heater and will be doing some testing over the next 24hrs.  Initially I did not get any errors in the klippy.log but last night it failed a print and the klippy.log pointed to a bad thermistor and/or heater.

jakevsnake, 2019-10-01:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3678631/klippy.log)
You can check out the klippy.log and the errors that were seen last night just as evidence to what a thermistor/heater failure looks like :)

KevinOConnor, 2019-10-19:
Okay, I'm going to close this as it looks like a solution was found for the original issue.

-Kevin

closed 2019-10-19 by KevinOConnor