Klipper3d/klipper #2028: MCU-specific controller fan?

theophile, 2019-09-30:
I've just set up a BIGTREETECH SKR Mini v1.1 as a secondary MCU in its own enclosure solely to drive four more extruders in a (probably ill advised) 6-in-1-out system. All six configured extruders use the same shared heater.

I've got a fan installed to cool the four TMC2208s on the Mini. The controller_fan function is very close to how I want it to behave, except that I only want the fan to come on when one of the four motor drivers on the Mini (secondary MCU) is activated. Right now it comes on whenever the extruder heater is activated (intended behavior of a controller_fan). I haven't checked yet but it might also come on whenever a stepper driver on the _main_ MCU is activated.

Is it currently possible to have this fan only be activated when one of the stepper drivers on the secondary MCU is activated, and then turn off after a few minutes past the last driver activity?

Hywelmartin, 2019-09-30:
As the extruders are all enabled together with the rest of the stepper a simple multipin isn't sufficient..
 maybe [THIS](https://github.com/KevinOConnor/klipper/blob/bbf6f87de05aca644c4e5b60283f81769f07917f/config/example-extras.cfg#L979) could work

theophile, 2019-10-02:
Thanks. Adding thermistors to the stepper drivers would probably be the most accurate solution. What we _really_ need is a way to read the driver's internal temperature directly and activate cooling as needed, like a CPU fan.

KevinOConnor, 2019-10-03:
Perhaps @wizhippo and/or @Master92 have some thoughts on this?

-Kevin

wizhippo, 2019-10-03:
It doe not support reading the actual temp from what I recall and quickly looked up.

Looks like you can get a`pre-warning` from `DRV_STATUS`.  Perhaps one could create a custom `sensor` that `temperature_fan` could use that reads this status and reports some fixed temp that would trigget it?

Otherwise you could also output the pre-warning to one of the diags pin, pass them through an `or` gate to a mosfet that turns the fan on accordingly.