Klipper3d/klipper #2029: MCU 'xyboard' shutdown: Move queue empty

richardvdweerd, 2019-10-01:
Hey,

I am new to klipper. Trying for several days now to get it running.
My printer is a piper2, corexy, with 4 z-motors. 
The 6 motion motors are connected. No extrusion motor yet.
The four z-motors are connected to a KBF2.0 board (x, y, z, e0 connectors resp.)
The x and y motors are connected to an Arduino UNO with a CNC-shield (x and y connectors resp.). The CNC-shield has one pin for enabling both motors. Is this an issue?
All motor drivers are TMC2100 in 1/4 steps in stealthChop. I used 1/16 steps first, but switched to 1/4 unload the micro-controllers.

Running a test with only x, y and z codes will put klipper in various  errors:
MCU 'xyboard' shutdown: Move queue empty
MCU 'xyboard' shutdown: Rescheduled timer in the past
MCU 'xyboard' shutdown: Invalid oid type
Lost communication with MCU 'xyboard'
MCU 'xyboard' shutdown: ?184
MCU 'xyboard' shutdown: Thermocouple ADC out of range

After a fresh install of OctoPi 0.16 and reconfiguring klipper the problem(s) persists.
It seems there is a problem with the xyboard. But what is it?
The connection? The Uno it self? or the cnc-board?

[klippy.log](https://github.com/KevinOConnor/klipper/files/3677706/klippy.log)

jalanjarosz, 2019-10-01:
Richard,
First, the UNO does not have enough "umph" to drive a coreXY via Klipper.    This has been tried on the Piper group to drive the 4 z-axis with no luck.   

Those of use in the Piper group have switched to putting the CNC Shield on a MEGA over the UNO.  The MEGA can handle it.

Feel free to PM me, or look on the Piper Facebook Group.  I have some basic instructions there on how to configure multi-MCU's for a Piper2.

Cheers
-James

richardvdweerd, 2019-10-01:
Hi James,

Thanks for the quick response. I wil look you up on facebook.

Richard

richardvdweerd, 2019-10-01:
Swapped the Uno for a Mega. Flashed the mega. Changed the serial-port in printer.cfg

In my test print file (no extrusion) I replaced all F5000 by F20000. With F5000 it came up to 5% of the print. With F20000 it's now at 10%. Looking good!!!!

It a real shake-down ;-). Some nuts came off...

Thanks James.

closed 2019-10-01 by richardvdweerd

KevinOConnor, 2019-10-01:
I'm curious why the Uno does not work properly.  The atmega328p is just as fast as the atmega2560 (actually, the 328 a little faster).

I haven't had a chance to analyze the log, but those errors generally indicate instability of the micro-controller - possible voltage issues, possibly a software defect, or possibly something else.  Are you sure the Uno you had was an atmega328p (as opposed to a 168 or some other chip)?

-Kevin

richardvdweerd, 2019-10-02:
My Uno has an Atmegz326P.
The replacement Arduino Mega2560 uses the same cables and is powered, like the Uno was, by the raspberry, using short usb cables

richardvdweerd, 2019-10-03:
As the 328P is slightly faster, I thought let's give it a try again. I installed a new Uno and tested it again. Running for 3 hours in a big test print without any errors.
Then I tried the first Uno again. After 6 minutes it ran into errors and in a second attempt just after 1 minute. I think the Uno is defective is some way.