Klipper3d/klipper #2031: Relation between [probe]z_offset, BED_MESH_CALIBRATE and SCREWS_TILT_CALCULATE?

andreklug, 2019-10-01:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3678276/klippy.log)

Hi, 

I experience that whenever I do a mesh calibration after correcting the screws tilt, the value set for z_offset in [probe] doesn't work anymore / has changed. From my understanding, the z_offset reflects the physical distance between the nozzle and the probe. This never changes on my printer, and several tests have shown that the probe accuracy is in the range of 0.0003mm. 

To make it simple: What's the correct order of steps / best practise to do a proper bed mesh leveling (after the bed tilt has been set)which then respects the z_offset?

Thank you!

Arksine, 2019-10-02:
The z_offset is not the z distance between the nozzle and the probe.  It is the distance between the nozzle and the bed when the probe triggers.  This is likely to change after manually adjusting the bed.

After doing a SCREWS_TILT_CALIBRATE, you should do a PROBE_CALIBRATE to set the correct z_offset.

andreklug, 2019-10-02:
thanks, and the Bed Mesh Leveling comes afterwards? I read in some issue that the SCREWS_TILT_CALIBRATE would discard the mesh results.

Arksine, 2019-10-02:
Yes, mesh leveling comes last.  My understanding of SCREWS_TILT_CALIBRATE is that it is simply a tool to help manually level the bed, it doesn't do any realtime correction...so it is compatible with bed_mesh.  The bed_tilt module is not, because like bed_mesh it does realtime correction on the Z axis.

Its possible to run BED_MESH_CALIBRATE before every print to get a fresh representation of the bed's geometry.  For some this isn't ideal because the probing process is slower, it just depends on your printer.

andreklug, 2019-10-02:
Thanks a lot for your help @Arksine !

The whole procedure really seems to be counterintuitive my point of view. Wouldn't it be way more logical to define the fixed distance between nozzle and sensor trigger point, like I thought it would be in the beginning? (I think this is also how it's handled in Marlin.)
Every time the printer is moved, the glas holding clamps are set to another position, or just anything that could impact the physics of the printer happens, the procedure has to be run again. With the value fixed in the way as Marlin does, a simple mesh calibration would do the job, assuming the distance is set once. 

What makes the procedure an annoyance is that the results seem to go off after a while. According to the documentation a "save_config" should do the job after PROBE_CALIBRATE, but after restarting the printer I often find the nozzle to be too close or away from the bed, and the whole thing starts again.

Thank you
andre

andreklug, 2019-10-02:
here's a quick impression of how I spent the last hours ;) 

[image]

by just changing the value on the 0.001 level, the printer goes from "here's you new air filter" to "let's see if we can press PLA into the glass finally", interchanging with "you ordered sand paper, sir?"

Arksine, 2019-10-02:
I don't know how marlin does it, but the relationship between the nozzle and the probe on the z-axis does not tell you where the bed is.  The only relationship that is important is the one between the nozzle and the bed when the probe triggers.  With that information you can define where exactly the nozzle is when the steppers are stopped.  Typically this value should be fixed, it would only change due to mechanical changes to the printer.

If a .001 difference in the z_offset is changing your layer height that much you likely have some kind of mechanical or probe accuracy issue.

andreklug, 2019-10-02:
The differences in the probe seem related to the bed heating turned on or off, I am currently separating the START GCODE in octoprint and in the slicer to make sure homing and probing are done before the bed heating is activated.

For the nozzle distance ... Reading your answer a few more times, we're maybe meaning the same. But I can't wrap my head around why changing the bed's Z position should have any effect then, because no matter where the bed is, the probe triggers always at the same (relative) Z distance, so Z=0 should be the surface of the bed - assuming it's flat. And for the sake of completeness, (on my printer) the probe is mounted to the extruder, not fixed to the bed.

In Marlin I understand it like this: Probe is triggered, z_offset equivalent says "hey nozzle, you'll find Z=0 now 2.134 mm underneath". Is it the same for Klipper?

"Typically this value should be fixed, it would only change due to mechanical changes to the printer." I agree with you if this means the distance between nozzle and probe, which then automatically defines the distance between nozzle and bed. But in this case I would never have to touch it again, except when I change one of the distances (nozzle - trigger; trigger - bed).

Thanks for your help and have a great - whatever it is where you are right now.

Arksine, 2019-10-03:
```
In Marlin I understand it like this: Probe is triggered, z_offset equivalent says "hey nozzle, you'll find Z=0 now 2.134 mm underneath". Is it the same for Klipper?
```

This is correct.  In this scenario, 2.134mm is the distance between the nozzle and the bed when the probe triggers.

In general, changing the bed's position should not change the z_offset, however given that you are having issues with layer height after manually leveling the bed I could only assume that it was somehow affecting the probe's trigger distance.   It sounds like you figured it out, noise from the heater could conceivably affect the probe as well.

andreklug, 2019-10-03:
Just came to my printer to figure that a Z axis problem is undeniable. I am impressed, it even managed to force-removed the nozzle.

[image]

johngianni, 2019-10-17:
I'm experiencing the same problems calibrating the z-offset and doing a bed level afterwards. Andreklug, have you solved your problems? Any particular suggestions about this matter? In my experience, it has become very difficult to calibrate the bed distance from the nozzle. I use, as you, a test gcode to print just one layer. I use the SET_GCODE_OFFSET Z=x to visually correct the z-offset. Once I think I've found the correct one, I save the value (adding or subtracting) to the relative [bltouch] parameter in the printer.cfg file. I issue a restart, but the new value seems not good while printing the same gcode. 
Questions:
-Do I have to set the SET_GCODE_OFFSET every time I print, rather then set the value in the printer.cfg file? 
-Do the mesh generated by the auto calibrate feature load after a restart or do I have to load it manually or worst do I have to do a mesh calibrate every time I print or start the printer (in the case there are no physical modifications to the printer? 
Thanks!

([bltouch]
........
x_offset: -47
y_offset: -8
speed: 5
z_offset = 3.525)

andreklug, 2019-10-18:
Hi John,

I am not sure I solved it. Sometimes the calibration seems to change, but I want to be sure I really exclude all other possible issues, so I am currently working to recreate the sensor mount with a stronger one and am rethinking it's position to make sure it doesn't change. 
Beside this, I took a few actions to remove other possible reasons. I am not using a BLtouch sensor but a proximity probe. As it's not working mechanically, I changed the preprint GCODE to heat bed and nozzle to a specific temperature, turn the heaters and fans off and then probe. Testing this takes a lot of time which I barely have at the moment.
Sometimes I also am under the impression that a new z_offset is not respected after changing it in the firmware and reloading the firmware, so I am shutting down the whole machine and start it again.
Also, I am not sure if / how the probe is taken into consideration with bed leveling / bed meshing. As I am using a very plane borosilicate glas I do test the leveling now and then, and sometimes the probe seems to change afterwards. But I am not getting to a point to prove it with numbers, it's a question of time.
So overall it's a bit frustrating but I am unable to point to a possible reason, and for sure I am not a professional when it comes to 3D printing.

How is your experience?

johngianni, 2019-10-21:
I'd like to write down in here my procedure and check with you if that's correct: 
G28
1) do a probe_calibrate with paper test and save_config after accept to store a z-offset
G28
2) do a level on all corners using the SCREW_TILT_CALCULATE.
G28
3) issue a BED_MESH_CLEAR
4) do a BED_MESH_CALIBRATE with 5,5 points 2 samples and SAVE_CONFIG
5) test print one layer to fine tune the z-offset. with SET_GCODE_OFFSET Z=x . Save the new value in the config file. RESTART

Latest outcome is that the print is good in the center (I have slightly curved Creality Stock Glass, resulting in a lower bed in the center) but squeezed on the sides. So it seems that it doesn't apply the mesh accordingly. 
Question: if I change the z-offset do I have to reissue a BED_MESH_CALIBRATE? 

Still struggling on it...

andreklug, 2019-10-30:
If this is the behavior, then it doesn't make sense. As I stated above, the offset between nozzle and probe usually never changes, and no one would expect to have to calibrate is if another geometry (bed tilt) is altered. 
Is there a reason why it has to be like this? Otherwise, it would be great if this can be changed.

KevinOConnor, 2019-11-06:
It's necessary to calibrate the probe in order to use it.  If it's calibrated against a bed with a severe angle, then it isn't going to have a good calibration.  It should, however, work fine when calibrated against a bed that is flat.

Best would be to use the BED_SCREWS_ADJUST mechanism (described at https://www.klipper3d.org/Manual_Level.html#adjusting-bed-leveling-screws ) to adjust the bed screws.  Since it does not rely on the probe, it does not have any issues with probe accuracy.

So, first level the bed, then calibrate the probe.

-Kevin

KevinOConnor, 2019-11-22:
I'm going to close this ticket as it appears the conversation has concluded.

-Kevin

closed 2019-11-22 by KevinOConnor