Klipper3d/klipper #2034: [Question] Is thermal runaway protection implemented in the MCU or on the pi?

diogosalazar, 2019-10-03:
Wondering if in the event the pi (or host computer) freezes/crashes/fails otherwise,  the MCU would be able to prevent a nozzle from melting on its own.

jakep82, 2019-10-03:
I don't know the specific answer to your question, but I do know that if the MCU loses connection with the host it will shut down.  In shutdown mode the heaters are disabled by default.

KevinOConnor, 2019-10-03:
There's info in the FAQ: https://www.klipper3d.org/FAQ.html#will-the-heaters-be-left-on-if-the-raspberry-pi-crashes

diogosalazar, 2019-10-10:
Thank you that is helpful

closed 2019-10-10 by diogosalazar

TwoToneEddy, 2019-11-15:
Sorry if this has been done to death but I can't find if Klipper has thermal runaway protection implemented? If so can someone point me to the commit that implements it please?
Thanks

jakep82, 2019-11-15:
@TwoToneEddy Yes, by default since the initial commit 3.5 years ago.

https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L199
https://github.com/KevinOConnor/klipper/blob/f582a36e4df16d5709943f7df17a900c8bcc12ab/config/example.cfg#L114

jakep82, 2019-11-15:
Also...

https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L744