Klipper3d/klipper #2035: [bed_mesh] Section yields error - not enough arguments for format string

radensb, 2019-10-03:
"Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: Internal error during connect: not enough arguments for format string"

This is the error I get when I include the [bed_mesh] section. Log file included.

I have a Lulzbot TAZ 6. I have gotten much to work with this firmware on that machine so far. The TAZ6 has a mechanical switch to home the Z axis. The nozzle is the bed level probe and touches a ground washer on each corner of the bed to ground the probe (nozzle). From what I can see, defining a probe creates a virtual endstop that is supposed to be used. But I technically have 2 Z sensors (a switch for Z homing, and the nozzle itself). Is there a way to configure the homing to use the switch (as it is now) and the bed-mesh to use the nozzle? That is the only thing that I can think of that is causing a problem with this.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3684610/klippy.log)

Arksine, 2019-10-03:
Bed Mesh requires a minimum 3x3 grid, so probe_count must be 3,3 or higher.  

You did however find a bug in the parse_pair function, it should have raised a more descriptive error telling you that the option needs to have a minimum value of 3.  I will fix it for the next release.

radensb, 2019-10-03:
Well, I'm glad this uncovered a bug! So, if 3x3 points are the minimum for Mesh, then that method is not correct for my printer. I will look into implementing the Bed Tilt method. The reason that I didn't think it was appropriate was that it wanted the user to configure the Z translations for X and Y. Isn't that the whole point of probing the corners? Anyway, I appreciate the guidance with Mesh.

Closing this as it was not being used properly by me and more descriptive error message should make invalid usage more clear.

closed 2019-10-03 by radensb

Arksine, 2019-10-03:
The x, y, and z adjust options don't need to be set, after BED_TILT_CALIBRATE is those options will be set internally.  If you have a fixed bed where you want to do a 1 time calibration you can use SAVE_CONFIG to save the bed_tilt settings to persistent storage or you can update them directly.  It appears that you want to calibrate prior to each print, so this would not apply to you.

If you don't get good results with all 4 corners, try using 3.

radensb, 2019-10-04:
I appreciate the post close followup. I enabled Bed_Tilt and performed a BED_TILT_CALIBRATE. It probed the 4 corners as expected. I then moved the head to position Z0 near each corner and measured 0.0mm - 0.1mm of Z height error. Seems to be working great!