Klipper3d/klipper #2037: Klipper v0.8.0 release

KevinOConnor, 2019-10-03:
This issue is notice that I am planning on making the next release of Klipper (v0.8.0).  My rough plan is to make the release in a couple of weeks (~20191020), with a feature freeze starting in a few days.

If anyone is aware of any critical bugs not already reported, please open a new github issue with a report of the issue.  With the start of the feature freeze, the Klipper master branch will focus on bug fixes and documentation updates.

-Kevin

arminth, 2019-10-04:
You didn't add a klippy.log!! ;-)

Good job!
Thx!
Cheers
Armin

saurabhceaseless, 2019-10-04:
Attaching klippy.log is an absolute must when raising any issue :-)

Great work! Keep it up!

Thanks
Saurabh

NickTom88, 2019-10-06:
Not sure if you would call this an issue but...

The TLDR version: the procedure for properly resetting the printer board to Stock. I'd like to set my Ender 3 board back to stock Marlin firmware. 

The short story: I purchased an SKR 1.3 w/ 2208v3.0 drivers. I'd like to take my Stock board and turn it into an easy backup board. I tried flashing it normally but it can't seem to connect to my computer. It's able to detect the port but that's as far as I can get.

Has a "printer factory reset" process been documented?

w1ebr, 2019-10-14:
What is your 0.8 integretion goals re: the scurve and pressure advance features?
Thanks!

w1ebr, 2019-10-21:
Hi Kevin! Will you announce 0.8 availability via this issue?

KevinOConnor, 2019-10-22:
The 0.8.0 version of Klipper has now been released!  For more information on the release, please see:
https://www.klipper3d.org/Releases.html

There is a Patreon page for those who want to donate to Klipper:
https://www.patreon.com/koconnor

New in this release:
* New G-Code command template support. G-Code in the config file is now evaluated with the Jinja2 template language.
* Improvements to Trinamic stepper drivers:
  * New support for TMC2209 and TMC5160 drivers.
  * Improved DUMP_TMC, SET_TMC_CURRENT, and INIT_TMC G-Code commands.
  * Improved support for TMC UART handling with an analog mux.
* Improved homing, probing, and bed leveling support:
  * New manual_probe, bed_screws, screws_tilt_adjust, skew_correction, safe_z_home modules added.
  * Enhanced multi-sample probing with median, average, and retry logic.
  * Improved documentation for BL-Touch, probe calibration, endstop calibration, delta calibration, sensorless homing, and endstop phase calibration.
  * Improved homing support on a large Z axis.
* Many Klipper micro-controller improvements:
  * Klipper ported to: SAM3X8C, SAM4S8C, SAMD51, STM32F042, STM32F4
  * New USB CDC driver implementations on SAM3X, SAM4, STM32F4.
  * Enhanced support for flashing Klipper over USB.
  * Software SPI support.
  * Greatly improved temperature filtering on the LPC176x.
  * Early output pin settings can be configured in the micro-controller.
* New website with the Klipper documentation: http://klipper3d.org/
  * Klipper now has a logo.
* Experimental support for polar and "cable winch" kinematics.
* The config file can now include other config files.
* Many additional modules added: board_pins, controller_fan, delayed_gcode, dotstar, filament_switch_sensor, firmware_retraction, gcode_arcs, gcode_button, heater_generic, manual_stepper, mcp4018, mcp4728, neopixel, pause_resume, respond, temperature_sensor, tsl1401cl_filament_width_sensor, tuning_tower
* Many additional commands added: RESTORE_GCODE_STATE, SAVE_GCODE_STATE, SET_GCODE_VARIABLE, SET_HEATER_TEMPERATURE, SET_IDLE_TIMEOUT, SET_TEMPERATURE_FAN_TARGET
* Several bug fixes and code cleanups.

For further information on Klipper, please see:
https://github.com/KevinOConnor/klipper

closed 2019-10-30 by KevinOConnor