Klipper3d/klipper #2038: Branch: SmoothPA - MCU 'mcu' shutdown: Stepper too far in past

hevilp, 2019-10-04:
I did the square_tower with a result of 0.725 PA (0.8 nozzle, 0.6 LH, lonjg bowden)
Now i make another test print and got the error:
MCU 'mcu' shutdown: Stepper too far in past

Running on Voron with Rpi3B + 2 x Skr 1.3 and tmc2208 with uart

[klippy.log](https://github.com/KevinOConnor/klipper/files/3689899/klippy.log)

hevilp, 2019-10-04:
I could fix until now, 16 microsteps on extruder instead of 32. Seems to be neccessary?

KevinOConnor, 2019-10-04:
Your pressure advance is quite high for that size nozzle - it'll be a challenge for the hardware to reach the motor rates to build the requested pressure.

The error is ultimately the result of trying to step near 250K steps per second on the extruder, which the micro-controller can not do with the default 2us step pulse time.  If you are exclusively using trinamic stepper motor drivers then you can select a step pulse time of zero in "make menuconfig" (under low-level options).  That will allow the micro-controller to issue more step pulses to a single stepper motor.  However, I suspect that the faster step time will just result in stripping of the filament.

Another option you could try is to tune the new `pressure_advance_smooth_time` - a higher smooth time will give the system more time to build pressure.  Details at https://github.com/KevinOConnor/klipper/pull/1997#issuecomment-538190663 .

-Kevin

closed 2019-10-15 by hevilp