Klipper3d/klipper #2039: [Feature Request] - Bed Mesh/Tilt Second Probe Speed

radensb, 2019-10-04:
Similar to second homing speed but applied to probes. This would allow for a quicker initial probe during bed leveling operation, followed by a slower, more accurate probe for each point.

jakep82, 2019-10-04:
https://github.com/KevinOConnor/klipper/pull/1356

KevinOConnor, 2019-10-19:
I'm going to close this as it looks like a dup of #1356 - if a developer wishes to implement and submit it then that's great.

-Kevin

closed 2019-10-19 by KevinOConnor

antst, 2019-10-24:
Just in case, I know little about python and nothing about clipper internals and design.
But I made short patch which adds extra G-code "BED_MESH_TILT". This code so far probe 3 points (in case of Cartesian it is (xmin,ymin), (xmax,ymin) and (xmid, ymax), in case of delta it is 3 points on edges which form equilateral triangle), and then from this probe it calculates vector which defines tilted plane (vector orthogonal to the plane). This vector can be used to transform ZMesh

https://github.com/antst/klipper/commit/8b902b21f90c06d71a8237ccc3e48f61d3c15867