Klipper3d/klipper #2040: [Feature Request] - Backlash Compensation

radensb, 2019-10-04:
Enabled in the config file. During a homing/bed mesh/bed tilt operation, once the sensor is TRIGGERED, the active motor would reverse direction and back off one step at a time until the sensor becomes OPEN. This would allow for the measurement of steps of the lost motion due to backlash in the axis when it changes direction. These steps would then be added to the axis movement on direction changes. This could also be applied to probing for more accurate Z backlash compensation.

jakep82, 2019-10-04:
This has been requested multiple times before, and Kevin has indicated it's not something he plans to implement.

https://github.com/KevinOConnor/klipper/issues/1166#issuecomment-458608061

KevinOConnor, 2019-10-19:
I'm going to close this as it looks like a dup of #1166 - it will require a developer to implement and submit it.

-Kevin

closed 2019-10-19 by KevinOConnor