Klipper3d/klipper #2041: provide a setup.py

mwalle, 2019-10-04:
Hi,

I'm trying to package klipper (the host part) with buildroot. The idea is to have a buildsystem which can generate a target image for the raspberry pi for example. 

I've seen issue #682 which provided a setup.py but was never merged. Is there any reason against a proper setup.py? For example the chelper is compiled with a custom python script, which does not work in cross compilation environment like buildroot (eg. the arm compiler runs on a x86 host), because the 'gcc' binary is hardcoded. Actually $CC should be used. But I guess that is only the tip of the iceberg. So having a proper setup.py which also compiles the extension modules will hopefully fix that.

Therefore, as a first step, I'd pick the setup.py from the mentioned issue. Fix it if needed and submit a pull request.

KevinOConnor, 2019-10-08:
This isn't something I've looked at.  I'm open to the idea if someone wants to implement and submit it.

-Kevin

mwalle, 2019-10-09:
Hi Kevin,

I've had a deeper look into that and there are some other issues which would need to be resolved first:

- [ ] remove the compile-on-time stuff
- [ ] rethink the hub ctrl calls. at the moment this is also in the chelper/__init__.py. but its not really a c helper but a "compile the binary if needed and call it with sudo".
- [ ] why is cffi used in the first place. wouldn't the proper way to make it a python c module, which can be imported and just called like any other python module? cffi is for foreign libraries which are not compiled by us.

-michael

KevinOConnor, 2019-10-19:
This isn't something I've looked at or plan to spend much time on myself.  The primary target is getting Klipper up on low-cost single-board-computers such as the raspberry pi.  If the code can also be packaged up without significant workflow changes to the primary use case, and someone wishes to tackle that, then that's great.

-Kevin

KevinOConnor, 2019-11-22:
I'm going to close this ticket as it appears the conversation has concluded.

-Kevin

closed 2019-11-22 by KevinOConnor