Klipper3d/klipper #2042: Fysetc cheetah 1.1 uart address mixed

mihero, 2019-10-04:
On my cheetah v1.1 board y and z axis tmc2209 drivers have mixed uart address.
While setting up sensorless homing I needed to switch so that y has uart_address 2 and z has uart_address 1. 
After switching addresses homing works great.

KevinOConnor, 2019-10-08:
A log would help here - or at least a copy of the config file that is correct.

-Kevin

mihero, 2019-10-08:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3703635/klippy.log)

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/3703637/printer.cfg.txt)

Sorry for missing files, was not sure if they would provide anything since after switching address homing started to work. This config is now from working version. Log contains few homing executions.

KevinOConnor, 2019-10-11:
That's odd - the schematic and Marlin both show uart address 1 for y and uart address 2 for z.

-Kevin

KevinOConnor, 2020-02-14:
Okay, I'm going to close this as it looks like the conversation has concluded.

-Kevin

closed 2020-02-14 by KevinOConnor

KevinOConnor, 2020-04-08:
Okay - I updated the default config (commit 4ea7b213).

-Kevin