Klipper3d/klipper #2043: Command Tuning_Tower Issue

thelastWallE, 2019-10-04:
Command Tuning_Tower is ending after only a few commands:

And it's starting with 0.514 but it should be 0.5?

[klippy.zip](https://github.com/KevinOConnor/klipper/files/3692425/klippy.zip)

KevinOConnor, 2019-10-05:
That could happen with some purge scripts - now with commit 6d7d079b that should be less likely.

The tuning value is based on the nozzle height - if the nozzle was at z=0.28 then it would generate a value of: `0.5 + 0.28 * .05 = .514`.

-Kevin

thelastWallE, 2019-10-05:
ah. can it be the primeline from cura? it is traveling up at the end and then to the modelprint.
I will test it again.
I have found a pythonscript in the meantime. But cura was limiting the wallspeed to 10mm/s because of the minimum layertime. So i must tune it again.

thelastWallE, 2019-10-09:
If i run the command the second time then it is aborting the tune_tower. but if i ran the command for the third time then it is also aborting the tune_tower. The receive message says so, i'm not sure if it is really aborting or just starting and gives me the wrong message back.

KevinOConnor, 2019-11-06:
As far as I know, this should work now.  If you're still running into an issue then please attach the Klipper log file from the session with the issue.

-Kevin

closed 2019-11-14 by KevinOConnor