Klipper3d/klipper #2044: M109 Blocking Commands not ACKing When Temp Reached

radensb, 2019-10-05:
I have tried to find other information on this but was unable.
There is an odd issue I am experiencing with the M109 commands that I just don't understand. M109 are supposed to be blocking "wait for temp" commands for extruders.
Here is a snippet from my start GCODE block taken from the original Lulzbot Start block:
```
M104 S{MATERIAL_SOFTEN_TEMPERATURE_0} T0  ; soften filament
M104 S{MATERIAL_SOFTEN_TEMPERATURE_1} T1  ; soften filament
M140 S{MATERIAL_BED_TEMPERATURE_LAYER_0}  ; get bed heating up
G28 X Y                      ; home X and Y
G1 X-17 F3000                ; clear X endstop
M117 Heating...              ; LCD status message
M109 S{MATERIAL_SOFTEN_TEMPERATURE_0} T0  ; wait for temp
M109 S{MATERIAL_SOFTEN_TEMPERATURE_1} T1  ; wait for temp
T0                           ; return to first extruder
G28 Z                        ; home Z
T0                           ; select this extruder first
```

What **should** happen is that both extruders get a temp set command to start heating up to the soften temperature, the bed starts heating, and XY are homed and we clear the switch. Then we wait for T0 to reach {temp - 5C}, followed by waiting for T1 to reach temp. Since M109 is blocking, we wait for T0 to ack before running the M109 command for T1. Since we ran the M104 commands for both extruders at the same time to the same temp, they heat up together so when the first  M109 command is reached, T1 is also ready when T0 acks, so the M109 command for T1 should just fall through.

What **actually** happens is that they both reach the target temp but something is still blocking. The Home Z doesn't command doesn't execute for another 30 seconds after the target temps are reached!? According to the Klipper documentation in the example-extras.cfg file, the default hysteresis is 5C - meaning that once the extruder is within 5C of the target temp, it is considered in range and M109 commands would get ack'd. But that doesn't seem to be happening...  This happens everywhere I run M109 commands.

I tried running the PID_CALIBRATE commands on the extruders. No change. I put thermocouples on the heater blocks and confirmed that the firmware is reading temp accurately within 2-3C. I added the `[verify_heater extruder]` section for each extruder and changed the hysteresis to 10C/20C/30C - nothing changed. I cant find much info on other or related issues, and I have to imagine that a code as common as M109 not working properly would be all over the place.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3693605/klippy.log)

FHeilmann, 2019-10-06:
The 5C is used for error_checks (think: heater_faults) and not to determine when the heatup sequence should end. 
The PID algorithm actually waits for the oscillations around the target temperature to settle before it returns.

radensb, 2019-10-06:
Ah, ok - well that explains why pulling that thread resulted in no operational difference.

With Marlin, the `M109` command was ack'd when the temp was within 10C of the target. This actually worked pretty well as it allowed the printer to continue while the final target temp was reached. This was especially nice on dual extruder prints that have a lot of toggeling between extruders because it saved a lot of time. Would it make sense to add a control for this behavior? Perhaps an addition to the `[extruder]` section with a `target_temp_in-range:` parameter that ack'd the M109 commands once the temperature was within ± `target_temp_in-range:`?

`M109 S215 T0` - ack'd at 205 when heating
`M109 S180 T0` - ack'd at 190 when cooling

radensb, 2019-10-11:
Does anyone see interest in this?

For a single extruder printer, I can see little to no penalty for the extended wait time to allow the extruder to settle, however, this becomes excessive in dual extruder configurations when the printer needs to toggle between the two extruders. The stock Marlin firmware would use the `M109` command to cool the inactive extruder down to the standby temp defined in Cura to prevent oozing. This worked very well. To save time, it proceeded past the `M109` command as soon at the temp heated up to within 10C or cooled down to within 10C of the target temp. As the print continued, the target would be reached and settle and since the target temp is typically well within the printable temp range for a material, the start of the extrusion would not be affected. This saves a significant amount of time on prints that have many extruder changes, like when printing with model and support material.

KevinOConnor, 2019-10-14:
FYI, this has been raised a few times before - for example, see #1108.  In short, I don't think it is a good idea to alter the behaviour of M109, but if someone wants to enhance SET_HEATER_TEMPERATURE to offer more control then that's fine.

-Kevin

radensb, 2019-10-15:
Thanks for the reply Kevin. This is mainly a dual extruder printer scenario. I did read issue #1108 previously, but didn't pay too much attention to it as it was referring to bed temps and I was not sure why you would need to have such control over the bed temp. In extruder change gcode scripts, I have found that it is common practice to place an `M109 R{material_standby_temperature}` command to then wait for the inactive extruder to cool down to the material standby temperature to prevent oozing and wiping on the next layers and this was acknowledged within 10C of the target temperature in Marlin. With the current operation of M109, this is painfully slow and with nozzle changes every other layer to support dual extrusion prints, the print time is stretched out dramatically. Since many people are coming to Klipper from Marlin, I think it would be great to allow this behavior so that existing profiles can be maintained with this regard. I agree that if this behavior is not standard (I only know Marlin behavior so it's standard to me) that it should be implemented as a config parameter (or perhaps an extended gcode command only?) that allows M109 to operate with an alternate behavior. I really wish I knew Python...

radensb, 2019-10-18:
Could this be done using a jinja gcode macro? Something like:

```
[gcode_macro wait_heat_first_extruder] 
gcode:
	{% for i in range(120) %}
		{%- if (printer.extruder0.temperature) > (printer.extruder0.target - 10) %}
                        {% break %}
                {% else %}
			G4 P1000
		{% endif %}
	{% endfor %}
```

So basically check if the extruder temp is greater than 10C less than the target and break, otherwise, delay 1 second and check again. Stop checking after 120 seconds. I pieced this together from examples on the referenced [jinja2 site](https://jinja.palletsprojects.com/en/2.10.x/templates/) but I cannot get it to work properly. It complains about the break statement. 

So, you could then do:
```
M104 S210 T0
wait_heat_first_extruder
START_PRINT ;<--- executes when T0 crosses 200C
```

radensb, 2019-10-22:
I changed the title of this as is as the settling behavior of Klipper seems to work properly, but needs to be addressed for dual extrusion prints that need more heater control.

None of my above experiments yielded any success. I tried some more dual extrusion test prints with a test model that requires supports. The results where very discouraging. Because of the excessive settling time for the active extruder to heat back up to the printing temperature (from its standby temperature) to actually start printing again, I was getting large ooze blobs that dragged all over the print. It also caused interference points that the head would crash into when moving over the blob.

I also noticed that the extended times caused the print to fail due to a heater issue detected by the firmware! When the fans come on to cool the inactive extruder, they stay on way too long because of the settling delay. Its like the heater dips too low and has to fight with the fan to eventually reach the settling temp. This also causes a conflict with the active extruder trying to heat up and in some cases causes the firmware to think something is wrong and throw the error because it is not heating up fast enough.

All my tests were done using CURA generated gcode and stock material profiles so it would seem that the Klipper M109 behavior is incompatible with dual extrusion printing that uses temperature control of the heaters to manage the materials. Kevin mentioned that he is didnt like the idea of altering the behavior of M109, but being that this is how CURA handles multiple extrusion material control (and other slicers as well), shouldn't this be considered a warranted change to enable the intended operation of the gcode from the slicers that Klipper was intended to support?

dingbatca, 2019-10-23:
I would love to see M109 have better support/compatibility.  As this has been brought up a few times, perhaps offer a range control to allow M109 to kick in sooner?  Thus you don't need to alter the default behavior of M109, but can allow for more control?

KevinOConnor, 2019-11-22:
I'm going to close this ticket as it appears the conversation has concluded.  As with the earlier #1108, if a developer wishes to implement and submit support for a more flexible SET_HEATER_TEMP type command then that's great.

-Kevin

closed 2019-11-22 by KevinOConnor