Klipper3d/klipper #2045: PID tuning / Job start temperature hyteresis

andreklug, 2019-10-06:
Hi,

I recently replaced the extruder heating element with a more powerful one (measured as 57 Watt) and did (several) PID tuning. My intention is to reduce waiting time before the print job starts. 
Actually, this thing fires like crazy - going from room temp to 220°C in a bit more than a minute. 

[image]

However, it always shoots 7 to 8 °C over the target temperature, then causing wait times of 4 to 5 minutes until it's back at for example 200°C. So now it's taking even more timeto start a print job.

I made sure the sensor is very close to the heating element to decrease delays in temperature changes.

a) Can I increase the tolerance before a print job starts? The hysteresis parameter in the [verify_heater extruder] section only seems to apply to checking the 

or 

b) Do you have any suggestions how to manually make the PID more precise? The current values are as follows, but I didn't find any explanation how they relate to the heating behavior.

#*# [extruder]
#*# control = pid
#*# pid_kp = 15.810
#*# pid_ki = 0.500
#*# pid_kd = 125.096

Thanks!
[klippy.log](https://github.com/KevinOConnor/klipper/files/3694460/klippy.log)

andreklug, 2019-10-07:
I found the reason for this behaviour.
The heater was connected through an SSR, and of course it doesn't do PWM. Lesson learned: SSR for heatbed is fine, for the nozzle it doesn't work well.

closed 2019-10-07 by andreklug

nophead, 2019-10-07:
You can use PWM with a bed heater with an SSR as long as it slow enough.
The minimum pulse width will be 10ms with a zero crossing SSR on 50Hz
mains, so with a 1 Hz PWM cycle you can have 1% resolution and is is still
fast enough that you won't see any thermal fluctuations.