Klipper3d/klipper #2047: Cannot start TMC 5160 driver

massasan, 2019-10-06:
Hello! 

Faced a problem of start of TMC5160 drivers from BIGTREETECH.

The initial data are as follows: firmware Klipper, slave controller Arduino Due on the Board RuRAMPS4D. nema17 Motors 2.5 A 24V power supply.

Drivers are interrogated by DUMP_TMC stepper= command, fulfill homing but at attempt to move axes are disconnected, thus alternately that X, U. it is possible to Include them again only reset of a power supply.

Tried put bloc nutrition on 12 V the situation has improved: driver work minutes 15 displacement occur eloquently without skipping steps. But then one of some sort of driver is disabled.

In both cases, the drivers continue to be polled by SPI!

Here is the crash driver dump

Send: DUMP_TMC stepper=stepper_x

Recv: // ========== Write-only registers ==========

Recv: / / TPWMTHRS: 0000005e TPWMTHRS=94

Recv: / / COOLCONF: 00000000

Recv: / / IHOLD_IRUN: 00061717 IHOLD=23 IRUN=23 IHOLDDELAY=6

Recv: / / TPOWERDOWN: 0000000a TPOWERDOWN=10

Recv: // ========== Queried registers ==========

Recv: // GCONF: 00000004 en_pwm_mode=1

Recv: / / CHOPCONF: 14410153 toff=3 hstrt=5 hend=2 tbl=2 tpfd=4 MRES=4 (16usteps) intpol=1

Recv: / / GSTAT: 00000005 reset=1 (reset) uv_cp=1 (Undervoltage!)

Recv: / / DRV_STATUS: 80174000 stealth=1 CSACTUAL=23 stst=1

Recv: / / FACTORY_CONF: 0000000c FACTORY_CONF=12

Recv: // IOIN: 30000052 REFR_DIR=1 DRV_ENN=1 AND SD_MODE=1 VERSION=0x30

Recv: / / LOST_STEPS: 00000000

Recv: / / MSCNT: 00000208 MSCNT=520

Recv: // MSCURACT: 010801f3 CUR_A=-13 CUR_B=it's 248

Recv: / / OTP_READ: 0000000c OTP_FCLKTRIM=12

Recv: / / PWMCONF: 00000000

Recv: / / PWM_SCALE: 0000006c PWM_SCALE_SUM=108

Recv: / / PWM_AUTO: 001a0091 PWM_OFS_AUTO=145 PWM_GRAD_AUTO=26

Recv: / / TSTEP: 000fffff TSTEP=1048575

Recv: ok

[...]

And this dump of a running driver

Send: DUMP_TMC stepper=stepper_y

Recv: // ========== Write-only registers ==========

Recv: / / TPWMTHRS: 0000005e TPWMTHRS=94

Recv: / / COOLCONF: 00000000

Recv: / / IHOLD_IRUN: 00061717 IHOLD=23 IRUN=23 IHOLDDELAY=6

Recv: / / TPOWERDOWN: 0000000a TPOWERDOWN=10

Recv: // ========== Queried registers ==========

Recv: // GCONF: 00000004 en_pwm_mode=1

Recv: / / CHOPCONF: 14410153 toff=3 hstrt=5 hend=2 tbl=2 tpfd=4 MRES=4 (16usteps) intpol=1

Recv: / / GSTAT: 00000005 reset=1 (reset) uv_cp=1 (Undervoltage!)

Recv: / / DRV_STATUS: 80174000 stealth=1 CSACTUAL=23 stst=1

Recv: / / FACTORY_CONF: 0000000b FACTORY_CONF=11

Recv: // IOIN: 30000052 REFR_DIR=1 DRV_ENN=1 AND SD_MODE=1 VERSION=0x30

Recv: / / LOST_STEPS: 00000000

Recv: / / MSCNT: 00000268 MSCNT=616

Recv: / / MSCURACT: 0139016c CUR_A=-148 CUR_B=-199

Recv: / / OTP_READ: 0000000b OTP_FCLKTRIM=11

Recv: / / PWMCONF: 00000000

Recv: / / PWM_SCALE: 00000076 PWM_SCALE_SUM=118

Recv: / / PWM_AUTO: 001b009e PWM_OFS_AUTO=158 PWM_GRAD_AUTO=27

Recv: / / TSTEP: 000fffff TSTEP=1048575

Recv: ok

Especially I have cause question " uv_cp=1(Undervoltage!)
"

Here is my config :

[stepper_x]

step_pin: PC5

dir_pin:!PC4

enable_pin:!PA7

step_distance:.01

endstop_pin: PC18

position_endstop: 0

position_max: 280

homing_speed: 50

second_homing_speed: 10

homing_positive_dir: false

homing_retract_dist: 3

[tmc5160 stepper_x]

cs_pin: PC6 # chip select pin of the SPI interface

spi_speed: 700000

microsteps: 16 # number of microsteps per full step of the motor

interpolate: true

run_current: 2.3 # value in amps

hold_current: 2

stealthchop_threshold: 150

# diag1_pin:^!PC18 # pin on the MCU where DIAG1 is connected (active low)

# driver_SGT: 60 # tuning value for sensorless homing, set to 0 as a start

[endstop_phase stepper_x] 

# The stepper_y section is used to describe the Y axis as well as the

# stepper motor controlling the X-Y movement.

[stepper_y]

step_pin: PD10

dir_pin:!PC3

enable_pin:!PA7

step_distance:.01

endstop_pin: PC17

position_endstop: 0

position_max: 310

homing_speed: 50

second_homing_speed: 10

homing_positive_dir: false

homing_retract_dist: 3

[tmc5160 stepper_y]

cs_pin: PC2 # chip select pin of the SPI interface

spi_speed: 700000

microsteps: 16 # number of microsteps per full step of the motor

interpolate: true

run_current: 2.3 # value in amps

hold_current: 2

stealthchop_threshold: 150

# diag1_pin:!PC17 # pin on the MCU where DIAG1 is connected (active low)

# driver_SGT: 60 # tuning value for sensorless homing, set to 0 as a start

[endstop_phase stepper_y]

Before that was TMC2130 everything worked perfectly except for the fact that the engines lacked power. 2130 not give more 1.2 A

Yes, I forgot to say. that under quick 5160 motry began to bask. Up to 50 degrees infrared thermometer. With 2130 on motrakh can be was the ice be preparing.

I suspect that the configuration of the default drivers which prescribes Klipper does not quite fit, but that's how they pravilno be configured to suit your needs don't know.

Can be who any with these drivers already understood.

Sorry in advance for my English, I have to use computer translation

massasan, 2019-10-07:
Here is my log file after driver error
[klippy.log](https://github.com/KevinOConnor/klipper/files/3696102/klippy.log)

KevinOConnor, 2019-10-08:
Do a web search for "tmc5160 datasheet" and download the pdf.  That document will have the information on how to tune the driver and what the error indicators mean.

-Kevin

closed 2019-10-08 by massasan