Klipper3d/klipper #2049: Action Commands Non-functional using M118 is [START_PRINT] Macro

radensb, 2019-10-06:
I have defined a set of Action Commands in Octoprint that control external fans through the Pi's GPIO interface. To use this with Klipper, I have added the `[respond]` section in the config file with `command` type. When I send: `M118 action:cool_E0_on` in the terminal, the fan spins up as expected. When I send `M118 action:cool_E0_off` in the terminal, the fan powers off.

However, when I run my [START_PRINT] macro (which has these M118 commands), I can see `Recv: // action:cool_E0_on` and `Recv: // action:cool_E1_on` in the terminal, as well as:
`21:07:56 action:cool_E0_on`
`action:cool_E1_on`
in the Klipper plugin Messages window.

And
`Recv: // action:cool_E0_off` and `Recv: // action:cool_E1_off` in the terminal, as well as:
`21:09:54 action:cool_E0_off`
`action:cool_E1_off`
in the Klipper plugin Messages window.

However, the fans never spin up. They seem to be ignored when being run from a macro!?

So, I tried running the gcode from the `[START_PRINT]` macro as descrete commands from a gcode file instead of calling the macro and the action commands worked as expected. So it seems that running the `M118` command from a macro fails? 

Is there some kind of limitation using commands or gcode in macros??

[klippy.log](https://github.com/KevinOConnor/klipper/files/3694850/klippy.log)

FHeilmann, 2019-10-07:
The difference between macros and sending the commands one by one is that octoprint views the macros as one command and continues after the whole start macro has been executed. It may be that octoprint does not parse action responses while it still waits for the conclusion of your print start macro.

radensb, 2019-10-11:
That makes sense. So would this be a limitation with Octoprint that is beyond the capability of the Klipper host software? I have noticed some some inconsistencies running other extended gcode commands within a macro. For example, trying to run the `SET_GCODE_OFFSET` extended command never seemed to work within the START macro, so I resorted to using the `M206` gcode command instead. Could this be related?

radensb, 2019-10-19:
Closing this as it seems to be an issue with how/when Octoprint processes responses and not with Klipper itself.

closed 2019-10-19 by radensb