Klipper3d/klipper #2050: G28 -> Internal error on command

TheRunnerPi, 2019-10-07:
Hi,

i sometimes get the Internal error on command:"G28" Error.

I saved two Klippy Logs short after the Error:

[klippy_05.10.2019.log](https://github.com/KevinOConnor/klipper/files/3696155/klippy_05.10.2019.log)
[klippy_06.10.2019.log](https://github.com/KevinOConnor/klipper/files/3696157/klippy_06.10.2019.log)

If i dont get the Error, my Z Axis moves way to quick going from G28 to G29. But i cant find the setting in the printer.cfg
G28 Homing ist done with the BlTouch centered in the middle. Then the Z-Axis kind of drops from 10mm to ~1mm and travels to the first probing spot. That "drop" movement should be slower. 

Regards,
Daniel

BlackStump, 2019-10-07:
first problem I see is your 128 microsteps
128 microsteps gives no better positional accuracy over 16 microsteps, what it does do is tax your mcu to beyond it limits.
try putting a G1 Z F300 in your G29 macro as it is trying to use F1500(25mm/s)

here is the error message in your log
>Internal error on command:"G28"
>MCU 'mcu' shutdown: Rescheduled timer in the past
>This generally occurs when the micro-controller has been
>requested to step at a rate higher than it is capable of
>obtaining. 

that would be related to your 128 microsteps try 16 instead

jakep82, 2019-10-07:
For reference, at your current microsteps you would need 192K steps per second to reach your max_velocity setting, and 80K steps per second to reach your max_z_velocity.  Your current MCU is simply not capable of achieving these step rates.

TheRunnerPi, 2019-10-08:
Tank you @BlackStump, i changed the microsteps and no more G28 Error.
G1 Z4 F300 slow movent to probing distance in the G29 macro, aslo worked :)

closed 2019-10-08 by TheRunnerPi