Klipper3d/klipper #2051: Strange delay when using OctoLapse

effernity, 2019-10-07:
Hello
I'm experiencing strange delays when using Klipper with OctoLapse plugin. I've opened issue on octolapse's github and with @FormerLurker we acknowledged in this  #https://github.com/FormerLurker/Octolapse/issues/387 that the delay is issued by M400 gcode

The klippy log attached, last print is that test gcode that @FormerLurker asked me to try, serial.log from octoprint: https://gist.github.com/effernity/1b6729b2627f57d652466249765153fc
[klippy.zip](https://github.com/KevinOConnor/klipper/files/3699393/klippy.zip)

Arksine, 2019-10-08:
Perhaps I'm missing something, but it seems to me that this is expected behavior.  The purpose of M400 is to wait until all moves are finished.  The lookahead buffer defaults to 2 seconds, so if the buffer is full there will be roughly a 2 second delay before M400 returns ok.

Put another way, if Octoplapse is looking to take a screenshot a specific printer position (hence the M114), then it would be required to wait that 2 seconds for the tool to reach the position matching M114.

FormerLurker, 2019-10-08:
The problem is that the delay is longer than expected, leaving the printhead motionless for over a second according to OP.

FormerLurker, 2019-10-08:
@Arksine, I've asked OP to run a test while using Octoprint (but with Octolapse disabled) while taking a video of the printer, and another test bypassing Octoprint completely, also while recording a video.  As I told @effernity, I don't want anyone going down any rabbit holes.

KevinOConnor, 2019-10-11:
I'm not sure I understand what is being reported here.  On an M400, it would be normal for Klipper to delay for at least 2 seconds.  Klipper normally keeps a lookahead buffer of around 2 seconds of moves.  Should a request come to wait for those moves to fully finish, that it will dutifully wait those ~2 seconds for all buffered moves to complete.

-Kevin

KevinOConnor, 2019-10-11:
To be clear though, Klipper can restart quickly after an M400, so it doesn't necessitate a long pause of the toolhead.  An M400 will result in a minimum toolhead delay of ~250-350ms - additional delays will depend on how quickly the software sending g-codes (eg, OctoPrint) can get moves to Klipper so that the lookahead buffer can be primed.

-Kevin

FormerLurker, 2019-10-12:
@KevinOConnor, the issue that was reported to me is the toolhead pause, not the fact that M400 waits for moves to finish as it should.  This user reported an approx 2 second of toolhead delay , and I've accounted for about 1185MS of this delay including an up to 350MS delay added by Klipper due to the M400.  However, after watching the videos posted on the other issue it looks like the delay is occurring at a different time than I expected, so I've asked the user to try some additional tests.

If Klipper is only adding 250-350MS delay, it can't be the problem.  However, please keep this issue open while we test some things on the other thread.  Thank you so much for your attention thus far!

FormerLurker, 2019-10-12:
OK, so I'm pretty sure this issue has nothing to do with Klipper.  I was convinced initially that it WAS klipper, especially since I've never seen such a problem running stock firmware with this printer, but it was a red herring.  I'm continuing to work with @effernity in the Octolapse issues, and will report back if I am wrong (again).

Thanks!

FormerLurker, 2019-10-12:
This issue has been resolved, though I'm not sure what was originally causing the problem.  I cannot close the issue myself, but I've verified with @effernity that this has been squared away.  Thanks for your time, and sorry to have asked for an issue to be created that was unrelated to Klipper (which is a VERY awesome project!  Thank you very much for creating/maintaining it!).

closed 2019-10-12 by effernity