Klipper3d/klipper #2052: SKR Mini E3 - V1.2 Driver changes

cfelicio, 2019-10-07:
Looks like the manufacture changed the schematics quite a bit, and the new release is not compatible with the previous firmware:

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/硬件/BTT%20SKR%20MINI%20E3%20V1.2/The%20Notement%20of%20BTT%20SKR%20MINI%20E3%20V1.2.pdf?fbclid=IwAR1hSVj5gWnTTs3cuC-ckgvypu1pDwV8Q0otCnkNOXyrTOgzCbFxjVBFChI

I have not received my board yet, but I suspect I will be getting V1.2. From a Klipper perspective, is the new release supported, or will require new development to get working?

cfelicio, 2019-10-07:
Here is the snipped of the change
[image]

KevinOConnor, 2019-10-08:
From your description, it seems like it would just require config file updates to specify the new pins.

-Kevin

epandi, 2019-10-08:
My v1.2 board is yet to arrive but I have tried to find all the differences in the source code of the BTT Marlin branch. Can you try my config file and report if this is working?

https://github.com/epandi/klipper/blob/epandi-SKR-mini-E3-1.2/config/generic-bigtreetech-skr-mini-e3-1.2.cfg

cfelicio, 2019-10-09:
Sadly I did not receive my board yet, I just saw the changelog on the forum. I'll reference this thread there and see if someone is using Klipper / willing to test right now. Thanks for going thru the changes and providing a sample :-)

Wireheadbe, 2019-10-14:
> @cfelicio
> My v1.2 board is yet to arrive but I have tried to find all the differences in the source code of the BTT Marlin branch (see below). Can you try my config file and report if this is working?
> 
> https://github.com/epandi/klipper/blob/epandi-SKR-mini-E3-1.2/config/generic-bigtreetech-skr-mini-e3-1.2.cfg
> 
> Marlin-2.0.x-SKR-Mini-E3-v1.2\Marlin\src\pins\stm32\pins_BTT_SKR_MINI_E3_V1.2.h
> #if HAS_TMC220x
> /**
> 
> * TMC2208/TMC2209 stepper drivers
>   */
>   //
>   // Software serial
>   //
>   #define X_SERIAL_TX_PIN  PB15
>   #define X_SERIAL_RX_PIN  PB15
> 
> #define Y_SERIAL_TX_PIN PC6
> #define Y_SERIAL_RX_PIN PC6
> 
> #define Z_SERIAL_TX_PIN PC10
> #define Z_SERIAL_RX_PIN PC10
> 
> #define E0_SERIAL_TX_PIN PC11
> #define E0_SERIAL_RX_PIN PC11
> #endif

Would that mean that e.g.
```
uart_pin: PC6
tx_pin: PC6
```

Should be configured with the same pin?

epandi, 2019-10-14:
Yes that means rx and tx is on the same pin.
The new config template was reported working in the discord channel so ive pushed a PR. 
#2069

Magoo0876, 2019-10-20:
Hi, I've tested what wireheadbe wrote, but Klipper does not like multiple times the same Pin in the config. The Uart alone is not enough.

KevinOConnor, 2019-10-30:
Thanks.  I'm going to close this as it looks like the issue was addressed with the merge of #2069 

-Kevin

closed 2019-10-30 by KevinOConnor

ice2009, 2020-03-01:
> Hi, Klipper does not like multiple times the same pin in the config by the way.
> 
> For the 1.2 version you have to disable the uart_adress and tx_pin, simply use the dedicated pin for each 2209.
> 
> Like that:
> 
> [tmc2209 stepper_x]
> uart_pin: PB15
> #tx_pin: PC15
> #uart_address: 0
> microsteps: 16
> run_current: 0.90
> hold_current: 0.500
> stealthchop_threshold: 250
> 
> PB15 XUART
> PC6 YUART
> PC10 ZUART
> PC11 EUART
> 
> ++++++++++++++++++++++++++++
> For the more advanced klipper users, the same with sensorless homing.
> Connect the diag pin of your board to a free endstop switch connector. As you may notice, mine is connected to the z switch because I use the PINDA2 on the servo connector.
> 
> [stepper_x]
> step_pin: PB13
> dir_pin: !PB12
> enable_pin: !PB14
> step_distance: .0125
> endstop_pin: tmc2209_stepper_x:virtual_endstop
> #endstop_pin: ^PC0
> position_endstop: 0
> position_max: 235
> homing_speed: 50
> homing_retract_dist: 0
> 
> [tmc2209 stepper_x]
> uart_pin: PB15
> diag_pin: PC2
> microsteps: 16
> run_current: 0.90
> hold_current: 0.5
> stealthchop_threshold: 250
> driver_SGTHRS: 140
> 
> And now become a PATREON and support Kevin. It is just fair if you use Klipper and all the support you get here!
> 
> have fun, Magoo

Hello Magoo you have config file for skr mini e3 with sensorless homing? Can you share please?