Klipper3d/klipper #2053: SKR v1.3 bed temp issue

rugalash, 2019-10-08:
I guess my board could be broken but I'll see if I am doing something wrong

I have many of the same temperature sensor, I have 3 plugged into TB TH0 and TH1 I've switched them around and tried different ones in different ports. It seems TH0 and TH1 always read fine around 22.6C currently and then my TB (bed) one reads like 6000C to -6000C and just flies around?

Not sure how to attach a file on this site and I don't think the log matters either, I restarted it to show the output:

`Stats 7977.0: gcodein=15 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=1017 bytes_read=4872 bytes_retransmit=9 bytes_invalid=0 send_seq=128 receive_seq=128 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99997090 heater_bed: target=0 temp=1997.1 pwm=0.000 print_time=1.933 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.3 pwm=0.000
Attempting MCU 'mcu' reset command
Restarting printer
Start printer at Mon Oct  7 21:15:30 2019 (1570500930.3 7978.6)
===== Config file =====
[stepper_x]
step_pin = P2.2
dir_pin = !P2.6
enable_pin = !P2.1
step_distance = .0125
endstop_pin = P1.29
position_endstop = 0
position_max = 320
homing_speed = 50

[stepper_y]
step_pin = P0.19
dir_pin = !P0.20
enable_pin = !P2.8
step_distance = .0125
endstop_pin = P1.27
position_endstop = 0
position_max = 300
homing_speed = 50

[stepper_z]
step_pin = P0.22
dir_pin = P2.11
enable_pin = !P0.21
step_distance = .0025
endstop_pin = P1.25
position_endstop = 0.5
position_max = 400

[extruder]
step_pin = P2.13
dir_pin = !P0.11
enable_pin = !P2.12
step_distance = .010526
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = P2.7
sensor_type = NTC 100K beta 3950
sensor_pin = P0.24
control = pid
pid_kp = 22.2
pid_ki = 1.08
pid_kd = 114
min_temp = 0
max_temp = 260

[heater_bed]
heater_pin = P2.5
sensor_type = NTC 100K beta 3950
sensor_pin = P0.23
control = pid
pid_kp = 22.2
pid_ki = 1.08
pid_kd = 114
min_temp = 0
max_temp = 130

[fan]
pin = P2.3
[... 248 lines cut ...]
Stats 8017.7: gcodein=40 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=1149 bytes_read=7337 bytes_retransmit=9 bytes_invalid=0 send_seq=150 receive_seq=150 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99997201 heater_bed: target=0 temp=3073.7 pwm=0.000 print_time=1.907 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.4 pwm=0.000
Stats 8018.7: gcodein=45 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=1155 bytes_read=7466 bytes_retransmit=9 bytes_invalid=0 send_seq=151 receive_seq=151 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99997207 heater_bed: target=0 temp=1528.8 pwm=0.000 print_time=1.907 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.1 pwm=0.000
Stats 8019.7: gcodein=45 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=1161 bytes_read=7566 bytes_retransmit=9 bytes_invalid=0 send_seq=152 receive_seq=152 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99997206 heater_bed: target=0 temp=7837.7 pwm=0.000 print_time=1.907 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.5 pwm=0.000
Stats 8020.7: gcodein=45 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=1167 bytes_read=7674 bytes_retransmit=9 bytes_invalid=0 send_seq=153 receive_seq=153 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99997208 heater_bed: target=0 temp=1710.5 pwm=0.000 print_time=1.907 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.3 pwm=0.000
Stats 8021.7: gcodein=45 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=1173 bytes_read=7781 bytes_retransmit=9 bytes_invalid=0 send_seq=154 receive_seq=154 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99997206 heater_bed: target=0 temp=1997.1 pwm=0.000 print_time=1.907 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.8 pwm=0.000
Stats 8022.7: gcodein=45 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=1179 bytes_read=7874 bytes_retransmit=9 bytes_invalid=0 send_seq=155 receive_seq=155 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99997203 heater_bed: target=0 temp=3073.7 pwm=0.000 print_time=1.907 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.6 pwm=0.000
Stats 8023.7: gcodein=50 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=1185 bytes_read=8001 bytes_retransmit=9 bytes_invalid=0 send_seq=156 receive_seq=156 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99997203 heater_bed: target=0 temp=1610.8 pwm=0.000 print_time=1.907 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.7 pwm=0.000
Stats 8024.7: gcodein=50 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=1191 bytes_read=8116 bytes_retransmit=9 bytes_invalid=0 send_seq=157 receive_seq=157 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99997202 heater_bed: target=0 temp=2541.5 pwm=0.000 print_time=1.907 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.5 pwm=0.000
Stats 8025.7: gcodein=50 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=1197 bytes_read=8216 bytes_retransmit=9 bytes_invalid=0 send_seq=158 receive_seq=158 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99997201 heater_bed: target=0 temp=4152.6 pwm=0.000 print_time=1.907 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.5 pwm=0.000
Stats 8026.7: gcodein=50 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=1203 bytes_read=8329 bytes_retransmit=9 bytes_invalid=0 send_seq=159 receive_seq=159 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99997202 heater_bed: target=0 temp=1710.5 pwm=0.000 print_time=1.907 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.3 pwm=0.000
Stats 8027.7: gcodein=50 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=1209 bytes_read=8444 bytes_retransmit=9 bytes_invalid=0 send_seq=160 receive_seq=160 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99997202 heater_bed: target=0 temp=3073.7 pwm=0.000 print_time=1.907 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.5 pwm=0.000
Stats 8028.7: gcodein=55 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=1215 bytes_read=8558 bytes_retransmit=9 bytes_invalid=0 send_seq=161 receive_seq=161 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99997200 heater_bed: target=0 temp=7837.7 pwm=0.000 print_time=1.907 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.7 pwm=0.000
Stats 8029.7: gcodein=55 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=1221 bytes_read=8671 bytes_retransmit=9 bytes_invalid=0 send_seq=162 receive_seq=162 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99997199 heater_bed: target=0 temp=1528.8 pwm=0.000 print_time=1.907 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.4 pwm=0.000
Stats 8030.7: gcodein=55 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=1227 bytes_read=8786 bytes_retransmit=9 bytes_invalid=0 send_seq=163 receive_seq=163 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99997200 heater_bed: target=0 temp=4152.6 pwm=0.000 print_time=1.907 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.4 pwm=0.000
Stats 8031.7: gcodein=55 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=1233 bytes_read=8886 bytes_retransmit=9 bytes_invalid=0 send_seq=164 receive_seq=164 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99997201 heater_bed: target=0 temp=7837.7 pwm=0.000 print_time=1.907 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.5 pwm=0.000
Stats 8032.7: gcodein=55 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=1239 bytes_read=8999 bytes_retransmit=9 bytes_invalid=0 send_seq=165 receive_seq=165 retransmit_seq=2 srtt=0.001 rttvar=0.000 rt [... cut ...] _time=1.907 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.3 pwm=0.000
Stats 8040.7: gcodein=65 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=1293 bytes_read=9914 bytes_retransmit=9 bytes_invalid=0 send_seq=174 receive_seq=174 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99997211 heater_bed: target=0 temp=7837.7 pwm=0.000 print_time=1.907 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.2 pwm=0.000
`

KevinOConnor, 2019-10-14:
Sounds like a hardware error or an incorrect pin was configured.  We use tools to analyze the logs so in the future please attach the log as described at https://www.klipper3d.org/Contact.html

-Kevin

closed 2019-10-15 by klipper-gitissuebot

rugalash, 2019-10-15:
> Sounds like a hardware error or an incorrect pin was configured. We use tools to analyze the logs so in the future please attach the log as described at https://www.klipper3d.org/Contact.html
> 
> -Kevin

Maybe because this account is brand new, or something is blocking it, either way it doesn't let me drag it or copy it. Anyway yea I'm pretty sure it's hardware, since I posted a week ago I've messed around with my multimeter on the board it seems different from the other temp pins so it's probably broken. 
Thanks