Klipper3d/klipper #2054: Inconsistent execution of gcode

joris57, 2019-10-08:
Following gcode snippet was executed twice consecutively 
```
G28  
G1 X0 Y0 Z5 F3000  
PROBE PROBE_SPEED=5 SAMPLES=3 SAMPLES_TOLERANCE_RETRIES=3 
SAMPLES_RESULT=median
G92 Z-0.4
G1 X0 Y0 Z150 F6000
set_gcode_offset z=50
G1 X0 Y0 Z10 F3000
```
In the first instance the nozzle ended up 60mm above the build plate as expected in position: 
```
get_position
// mcu: stepper_a:4803 stepper_b:4802 stepper_c:4801
// stepper: stepper_a:291.450033 stepper_b:291.444831 stepper_c:291.447968
// kinematic: X:-0.005145 Y:0.000593 Z:60.682796
// toolhead: X:0.000000 Y:0.000000 Z:60.682796 E:0.000000
// gcode: X:0.000000 Y:0.000000 Z:60.682796 E:0.000000
// gcode base: X:0.000000 Y:0.000000 Z:50.682796 E:0.000000
// gcode homing: X:0.000000 Y:0.000000 Z:50.000000
```
In the second instance, the nozzle ended up 10mm above the build plate in position:
```
get_position
// mcu: stepper_a:803 stepper_b:802 stepper_c:801
// stepper: stepper_a:241.450033 stepper_b:241.444831 stepper_c:241.447968
// kinematic: X:-0.005145 Y:0.000593 Z:10.682796
// toolhead: X:0.000000 Y:0.000000 Z:10.682796 E:0.000000
// gcode: X:0.000000 Y:0.000000 Z:10.682796 E:0.000000
// gcode base: X:0.000000 Y:0.000000 Z:0.682796 E:0.000000
// gcode homing: X:0.000000 Y:0.000000 Z:50.000000
```
I would have expected the same result for both executions of the code.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3702437/klippy.log)

andreklug, 2019-10-08:
is it the same when you add an G90 (set to absolute position) in front of your code?

joris57, 2019-10-08:
No difference with or without G90.

pasiz, 2019-10-09:
Your code doesn't reset Z offset, so in second case you have z offset +50 before executing G1 X0 Y0 Z150 F6000. I suggest that you reset your z offset before running that.

joris57, 2019-10-09:
You are correct that that would fix the code. The issue, however, is not how to work around the problem, but whether Klipper functions here as intended? In other words : with what logic could I have predicted the different outcomes in running this code?

Looking further into it, I noticed that the gcode offset is not applied to the G92 Z-0.4 command.

I leave it up to the developers to comment if this is a feature or a bug. :-)

KevinOConnor, 2019-10-11:
The result looks correct to me.  The SET_GCODE_OFFSET command is intended to remain in effect after a G28 (while G92 is intended to not remain in effect).  On the second G92 invocation you instructed Klipper to treat the current position (~Z=-50) as the origin, the subsequent moves are then relative to that origin.  The second SET_GCODE_OFFSET command has no effect as an offset of Z=50 was already in place.

More generally, though, it's not productive to mix these commands in odd ways like the above - I recommend avoiding that.

-Kevin

joris57, 2019-10-11:
Two observations : 

I believe the most widely accepted definition of origin (in geometry) is the point with coordinates all zero. E.g. (0,0) in two dimensions and (0,0,0) in three. 'To treat Z=-50 as the origin' is, in my opinion, contradictory to this definition and certainly the source of my confusion. My expectation would be that G92 Z=0 DEFINES the current position as the (in this case Z) origin (i.e. resets the coordinate system so that Z=0 at the current position). 

With all due respect (which I try to show by being a patreon to Klipper), but how is an end-user to know what is or is not 'odd ways' for Klipper?

KevinOConnor, 2019-10-11:
I don't think I did a good job of explaining what occurred.  Your test case simplified looks like this:
```
G28                     ; Home the axis
G92 Z0                  ; Make future g-code moves relative to the current position
G1 Z150 F6000           ; Move up 150mm from the last G92Z0 request
SET_GCODE_OFFSET Z=50   ; Perform a virtual shift of the Z axis
G1 Z0 F3000             ; Move down 100mm (back to last G92Z0 + 50mm z axis shift)

G28                     ; Home the axis
G92 Z0                  ; Make future g-code moves relative to the current position
G1 Z150 F6000           ; Move up 150mm from the last G92Z0 request
SET_GCODE_OFFSET Z=50   ; Does nothing as virtual axis shift already in place
G1 Z0 F3000             ; Move down 150mm (back to the last G92Z0 request)
```

FWIW, I find the behaviour of G92 and other traditional g-codes to be entirely baffling.  I'm at a loss for how to describe them in a meaningful way.  Alas, the above G92 behaviour is required in order for Klipper to be compatible with standard slicers and 3d printer interfaces.

-Kevin

BlackStump, 2019-10-11:
https://reprap.org/wiki/G-code

joris57, 2019-10-14:
Thanks. I see now where our views of 'offset' and G92 Z0 differ.

For me, an offset is something that gets added to the movement vectors 'on the fly'. For you, it is a translation of the origin, which, most of the time, yields the same results, except when the offset changes, where, with your logic the origin moves.

When probing for the build plate, followed by G92 Z0, I would have expected the build plate to be at Z=0 for a Z-offset=0, regardless the value of the offset at the moment of probing.

Anyway, the Klipper logic is what it is and at least it is clear to me now, so I'm closing he issue.

closed 2019-10-14 by joris57