Klipper3d/klipper #2055: PSU Control, load Config after StartUp

TheRunnerPi, 2019-10-09:
Hey there :)

i did install a Relay Board to power down the printer when idle.

If have got a 12V Power Supply which is switched On and Off by the Relay.
A 5V Power Supply is continuously On and powers the Raspberry PI. 
The Raspberry controls the Relay Board via the "PSU Control" PlugIn.

If i do a M80, the 12V Power Supply switches On and gives the RAMPS 1.6 Board power.
Problem seems to be that the TMC2208 dont load the config, after powering up the Board. 
When i start a Print or do a G28, the Steppers drive to fast and crash at the End.  

If i do the M80 and a Restart, everything is okay.

Is there maybe a way to load the config in the startup gcode script? 
Because a Restart kills the current print..

TheRunnerPi, 2019-10-09:
Klippy of the Issue
[klippy_09.10.2019.log](https://github.com/KevinOConnor/klipper/files/3707251/klippy_09.10.2019.log)

FHeilmann, 2019-10-09:
Use the `INIT_TMC` command described [here](https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#tmc2130-tmc2660-tmc2208-tmc2209-and-tmc5160) to explicitly initialize the TMC after you have powered them on. TMC need VIN to initialize properly, and since your MCU probably boots up (and tries to initialize the drivers) before VIN power is applied that part fails.

TheRunnerPi, 2019-10-09:
```
M80
G4 P3000
INIT_TMC STEPPER=stepper_x
INIT_TMC STEPPER=stepper_y
INIT_TMC STEPPER=stepper_z
```
Works in the starting Gcode Script :) 
Now if i want to wake up the printer via GCode, the issue still occurs.
[image]

Could there also be a Workaround for that Problem?

TheRunnerPi, 2019-10-12:
Other Question to find a solution, can i modify the M80 command?

Like i do with the G29:
```
[gcode_macro G29]
gcode:
	G1 Z4 F300
	BED_MESH_CALIBRATE
	BED_MESH_OUTPUT
```
Maybe like this:
```
[gcode_macro M80]
gcode:
        M80
	INIT_TMC STEPPER=stepper_x
        INIT_TMC STEPPER=stepper_y
        INIT_TMC STEPPER=stepper_z
```
Regards,
Daniel

KevinOConnor, 2019-10-19:
This looks like a dup of issue #1702.

-Kevin