Klipper3d/klipper #2058: Nozzle Scraping after 1st layer

tmaniac, 2019-10-10:
Hello,

i've been running Klipper on my CoreXY converted system.
However, recently i notice that the nozzle is scraping (really bad) over the previous layers.
It makes a gigantic noise and i can't imagine this is very good for the print/printer

The first layer is put down perfectly, the scraping starts from the 2nd layer.

I double checked my slicer settings (PrusaSlicer 2.1) and even tried Cura.
From both slicers i get the same results.

I tried with PA enabled and PA disabled to see if that made any difference.

I did the bed-level paper-test multiple times.
The steps for my Z-axis are correct; i copied the original manufacturers marlin step value (400) and recalculated them to Klipper-format (1/400=0.0025)

Some hardware info:
MCU: MKS Base Gen L
Stepper drivers: TMC2208 UART
My Z-axis consists of 2 stepper motors+leadscrews connected to a splitter. (FLSUN Cube default)
BLtouch clone (3dtouch)
BMG extruder clone (TriangleLabs)
Standard FLSUN Hotend
Steel nozzle

Print temp: 215c
Bed temp: 55c

I'm a bit at a loss as to where to check/what to try.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3712377/klippy.log)

FHeilmann, 2019-10-10:
most slicers don't retract when crossing over the first layer. So if your first layer is a smidge too close or you're overextruding the nozzle can drag through the first layer.

Here's the 2 things I would check:

- Irrespective of the paper test, inspect your first layer and see if it is too squished or not
- Calibrate your extrusion (extruder steps/mm + extrusion multiplier for the given filament)

BlackStump, 2019-10-11:
I would also add check that your Z steps are in fact correct by commanding Z to move a 150mm (as a example) and measuring that distance is should be precisely 150mm

tmaniac, 2019-10-11:
i will check if i can re-calibrate the extrusion, perhaps that will solve my problem...

btw, to give you an idea as to what's happening; here is a video from a print last night. (tip: sound on ;) )
[https://photos.app.goo.gl/wNCQ6sdYdiWhDA4m6](url)

BlackStump, 2019-10-11:
that is a interesting sound, are you sure it is the nozzle scraping and not a mechanical issue else where?
ie idlers, bearings.
try removing your filament and printing with out filament ie dry run

tmaniac, 2019-10-11:
Will try a dry-print tonight;

Pretty sure though, it only happens when printing solid layers.
When doing perimiters or infill it's barely audible.

If i do free moves over x or y (fast or slow) i don't have that noise.

At first i thought, hey! perhaps i should do the paper test with everything heated to printing-temp!
But that did nothing (the opposite actually, nozzle was way to far from the bed when i started printing)

theopensourcerer, 2019-10-11:
FWIW, that doesn't sound like a hot nozzle scraping through molten plastic to me.

nophead, 2019-10-11:
Just sounds like classic over extrusion to me. When laying down a solid layer the plastic is molten, so the nozzle will plough through it, but when it goes back over it the plastic has hardened, so it scrapes on the ridges. On layers that are not solid there is room for the plastic to expand sideways, so it doesn't create high spots that scrape.

BlackStump, 2019-10-11:
if it is over extrusion then usually that is easy to see on the solid layers as @nophead has mentioned, if the print has no ridges ie printing clean and smooth then it clearly isn't nozzle scraping.
cancel the print when it is making the sound and visually inspect the print surface.

tmaniac, 2019-10-15:
ok, things i've done:
- Print without extruding: no scraping sounds
- calibrated extrusion multiplier (when printing 20x20 cube i measured 20.40mm so i recalculated multiplier to 0.95)
- re-calibrated probe using paper-test

first layer looks good, not too squished.
the (solid) layers after that keep scraping.

still a weird issue.
Maybe i should start with a virgin config file.

KevinOConnor, 2019-10-30:
Any further updates on this?  This doesn't seem like a Klipper issue.

-Kevin

tmaniac, 2019-11-02:
Sorry for the late reply.
I think there is more wrong with my hardware. 
Currently experiencing freezes of 1 ~ 2 seconds mid print... I think my mks base gen L is dieing on me. 

Planning to replace the board with an skr pro v1.1 with tmc2209.... 

Just waiting for the 11-11 sales to start :p

KevinOConnor, 2019-11-14:
Okay, it sounds like the conversation has concluded on this topic then.

-Kevin

closed 2019-11-14 by KevinOConnor