Klipper3d/klipper #2059: [Feature Request] Add additional heaters support

aegelsky, 2019-10-10:
Hi Kevin and all appreciated contributors.

Is it possible to add additional heaters the way heatbed works now?
Could not find it in config or googling for Klipper.
Also adding "virtual" extruders is a bad option imho, it will need several pins added for each one, which could be used other way (just thinking..).
Many firmwares support it, like extra heater(s) for closed print box.
In my case I need three more heating beds.
I have one now for testing (550x1230mm) and it will be three more, to make 1.23x2.2 meters array.
Also might be couple extra heaters for enclosed print area (and many people already have them now, to keep nice working temperature in the box.

How hard is it to make heaters handling like you already have with additional steppers?

I don't like python much and not sure it will be best if I do it, didn't code anything big for several years by now.

Best regards,
Alexander.

FHeilmann, 2019-10-10:
have a look at the multi-pin extra. It basically allows you to control multiple pins (i.e. your heater pins) as if they were a single pin.
See here: https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L1176

KevinOConnor, 2019-10-11:
Also look at the generic_heater support - described in config/example-extras.cfg.

-Kevin

aegelsky, 2019-10-14:
Thanks! I'll give it a try and write back.
Or just close this issue?

closed 2019-10-30 by KevinOConnor