Klipper3d/klipper #2062: Filament Sensor and IDLE Time

schnello, 2019-10-12:
Hello.

I want use the fila sensor:
The target is:
Fila Sensor detect filament runout >>>
Hotend off
Bed stays on
Motors stays on

Currently Klipper turns off the motor and the Bed after i guess 10 min.

I found now the possibility to set the timeout:
[idle_timeout]
gcode:
 M84
timeout: 18000

My question:
If i set the timeout... will the bed continue to be heated?
Any other option to set the timeout to a higher value only if the filament sensor is active? 
Basically I did not like to change the global timeout
Feature Request:
Filament Sensor should set the timeout

[klippy.log](https://github.com/KevinOConnor/klipper/files/3720277/klippy.log)

FHeilmann, 2019-10-12:
There's a gcode SET_IDLE_TIMEOUT that does exactly what you want. Check the docs.
The bed stays heated unless commanded otherwise by gcode.

schnello, 2019-10-12:
Thanks a lot!

Maybe it make sense to add this information to the filament runout section. A filament runout sensor with the default idle make no sense.

closed 2019-10-12 by schnello