Klipper3d/klipper #2063: Getting 'Heater extruder not heating at expected rate' in the same print point.

andrico21, 2019-10-12:
Got 'Heater extruder not heating at expected rate' error on previous print, then resliced the same STL with some additional parts added - print job failed again on the same height/point (about 7hrs progress). Also after Klipper was shut down I noticed the chassis FAN is active, not sure - how to check the fact it was run sometime during print with klippy.log
[image]

[klippy.log](https://github.com/KevinOConnor/klipper/files/3720926/klippy.log)

tonn0297, 2019-10-12:
looks like you have a layerfan or some cooing on your heeterblock. it goes from 252 to 243 in a short amount of time

andrico21, 2019-10-12:
@tonn0297 thanks for idea! thought I turned off the cooling, but now I've looked on this layer in slicer and realized it's bridging layer - first one after infill. Then I've checked cooling and discovered the autocooling is enabled.
Solved!

closed 2019-10-12 by andrico21