Klipper3d/klipper #2065: SAVE_CONFIG wiped my configuration file

johannes-majamaki, 2019-10-13:
I doubt this issue is of a nature where a log file will be especially meaningful, but I shall gladly add one if it turns out so.

Chain of events: I am setting up my printer after a major rebuild, and level my bed with the manual bed levelling utility. I realise I hadn't wired up my hotend fan, and thus proceed to turn off my printer - with an underlying thinking process that I had just performed a procedure with a helper script and I would want to save its results: not remembering that the manual bed levelling utility does not produce any config results. SAVE_CONFIG, power off, and off to connect some cables.
I turn it back on to print out a calibration cube, and I find my entire clipper configuration file is blank. Unless the system creates a backup somewhere, looks like I am back to square one. Do tell me if it does, since I am very much going to leave this for tomorrow after this setback!

Anyway, as to the core of this issue: I hope you will consider adding some safety feature, since (unless this issue is about something else) SAVE_CONFIG is actually a pretty dangerous command to run out of its correct context.

All the best,
Johannes

Hywelmartin, 2019-10-13:
The save_config makes a backup in /home/pi/

johannes-majamaki, 2019-10-13:
> The save_config makes a backup in /home/pi/

Thank you, thank you, thank you.
My board is a tad weird - I mean, not horrifically, but nonetheless - so I was fully expecting a couple of hours of work. (It's funny how the SCP thing seems so much easier to figure out when one is more motivated xD). Now I am back in business!

I think the argument could be made that SAVE_CONFIG should check if the version of the file it intends to save is 0kb unless it going blank is used somewhere. Nonetheless, this version history is a sweet sweet thing. Thank you once again!

closed 2019-10-13 by johannes-majamaki