Klipper3d/klipper #2066: [question] Use of `Step pulse duration` work workaround TMC 2208s in standalone mode not working with clipper

mattkenn4545, 2019-10-13:
I currently have 'silent' Creality 1.1.5 board for my ender3 that I'd love to install however it has 2208s and they don't work well with Klipper.

In the MCU make config there is an option to set `Step pulse duration`.  Maybe this would avoid the issue there klipper is too fast for those drivers in standalone mode?

KevinOConnor, 2019-10-14:
It's not valid to use the tmc2208 in standalone mode with Klipper.  They work fine in uart mode.  They can cause seemingly random failures in standalone mode - as described at: https://www.klipper3d.org/FAQ.html#my-tmc-motor-driver-turns-off-in-the-middle-of-a-print

There is no workaround for the tmc2208 in standalone mode.  Neither pulse duration, nor smoothing, nor selection of kinematics provide any assurance that the issue will not occur.

-Kevin

mattkenn4545, 2019-10-15:
That is what I thought. Have a board that works 'perfectly' with the drivers that would quench the whine that I can't really use.

Wish there were some way to workaround the issue with klipper.

Closing and thanks

closed 2019-10-15 by mattkenn4545