Klipper3d/klipper #2068: Reverse encoder only für menu (not for values)

schnello, 2019-10-13:
Hello

I use the Reprap Smart Controller and the encoder is in the right side.
Is there a config option to reverse the encoder only for the menu navigation? If i swap the encoder pins the navigation is correct (left > down, right > up) but from this point the values are wrong (left > increase, right > decrease)
[klippy.log](https://github.com/KevinOConnor/klipper/files/3722482/klippy.log)

KevinOConnor, 2019-10-19:
As indicated, it's possible to reverse the direction by swapping the pins.  It is not possible to use one direction for menu navigation and another direction for greater/less than.

-Kevin

schnello, 2019-10-20:
Thank you Kevin. Any change to implement this in the future? I guess most of the LCD panels have the encoder on the right side.

KevinOConnor, 2019-11-22:
I'm going to mark this as inactive as it doesn't seem like anyone is currently working on implementing this.

-Kevin

closed 2019-11-22 by KevinOConnor