Klipper3d/klipper #2071: Error when homing from more than 200mm

aegelsky, 2019-10-14:
Got this error today:
Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: MCU 'zboard' shutdown: Timer too close

I remind that I have six drv8825 drivers connected to 6 Z axis motors of my big boy (Z = 2000).
They work with only middle step jumper set (1/4 step, they lacked torque at 1/32 and it was another error long before when homed Z before, which I bypassed by 1/4 stepping, not sure if I need it still).

It still works on ramps/mega, might worth switching to SKR?
Does it worth investigating anyway?
Sometimes I get this error when I home from > 200mm (400-500-600), sometimes it goes well from 600+.
Not a big deal if you have more important stuff, it work most of the times, like 80% of attempts.

Attached klippy.zip, it has klippy.7z inside and it has klippy.log inside.
Souvenir from Russia - matrioshka (nesting doll) :))
Just github didn't want to attach 7z.
[klippy.zip](https://github.com/KevinOConnor/klipper/files/3725565/klippy.zip)

aegelsky, 2019-10-14:
Got unexpected same error two times in a row when was homing from 0, i.e. triggered Z.
I use probe btw.

[klippy.zip](https://github.com/KevinOConnor/klipper/files/3725645/klippy.zip)

Trying to continue failed print with start gcode like this:

`M82 ;absolute extrusion mode
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops

G1 Z269.6; LAYER HEIGHT! #### !!!!!

M104 S210
M105
M109 S210

G92 E1991.15507 ;current extruded length #### !!!!!
G92 E1991.15507 ;current extruded length again #### !!!!!

G1 F9000
;Put printing message on LCD screen
M117 Print shit...

G92 E1991.15507 ;current extruded length #### !!!!!
G92 E1991.15507 ;current extruded length #### !!!!!

M107
M106 S255

G0 F1500 X161.697 Y229.255 Z269.6
G0 F7200 X157.99 Y231.641
;TIME_ELAPSED:11821.800852
;LAYER:538
;TYPE:WALL-OUTER
G1 F1500 E1991.15507
G1 F4500 X158.022 Y231.618 E1991.16195
G1 X158.044 Y231.603 E1991.1666
G1 X158.093 Y231.57 E1991.17691`

KevinOConnor, 2019-10-14:
Looks like a dup of #1460.  See that issue for workarounds.  I have a rough plan to address it, but no timeframe to complete that work.

-Kevin

closed 2019-10-30 by KevinOConnor