Klipper3d/klipper #2072: SKR E3 DIP with TMC5160 (pin PB15 is reserved for spi2)

r15ch13, 2019-10-14:
I was trying to use the `SKR E3 DIP` with TMC5160 drivers.
I copied the `generic-bigtreetech-skr-mini-e3.cfg` and changed the PINs according to the [sheet](https://cdn.shopify.com/s/files/1/1619/4791/files/BIGTREETECH-SKR-E3-DIP-V1_06.jpg) from BIGTREETECH and also referencing the [pins_BIGTREE_SKR_E3_DIP.h](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/stm32/pins_BIGTREE_SKR_E3_DIP.h) from Marlin 2.x.

But after running `sudo service klipper restart` it shows the following error:
`pin PB15 is reserved for spi2`

I then change it to another PIN (for testing) and got the same error for `PB14` and `PB13`.
According to https://www.klipper3d.org/stm32f1.html these PINs are used for SPI or as general I/O.

Digging around the source revealed: https://github.com/KevinOConnor/klipper/blob/1e8582e3f6cb13d85f06c740d7f79824c15a88be/src/stm32/spi.c#L17-L18
Is it possible to disable `spi2`?

[klippy.log](https://github.com/KevinOConnor/klipper/files/3725895/klippy.log)

r15ch13, 2019-10-14:
Setting `spi_bus` to `spi1` seems to work.

```ini
[tmc5160 stepper_x]
cs_pin = PC10
spi_bus = spi1
```

Running `STEPPER_BUZZ STEPPER=stepper_x` does nothing though...

[klippy.log](https://github.com/KevinOConnor/klipper/files/3726116/klippy.log)

trevor-powdrell, 2019-10-15:
I setup the E3 dip with tmc2130 in spi mode. 
I used SPI3 pins

[tmc2130 stepper_x]
cs_pin: PC10
#   The pin corresponding to the TMC2130 chip select line. This pin
#   will be set to low at the start of SPI messages and raised to high
#   after the message completes. This parameter must be provided.
#spi_bus: SPI3
#spi_speed:
spi_software_sclk_pin: PB3
spi_software_mosi_pin: PB5
spi_software_miso_pin: PB4

[tmc2130 stepper_y]
cs_pin: PC11
#   The pin corresponding to the TMC2130 chip select line. This pin
#   will be set to low at the start of SPI messages and raised to high
#   after the message completes. This parameter must be provided.
#spi_bus: SPI3
#spi_speed:
spi_software_sclk_pin: PB3
spi_software_mosi_pin: PB5
spi_software_miso_pin: PB4

[tmc2130 stepper_z]
cs_pin: PC12
#   The pin corresponding to the TMC2130 chip select line. This pin
#   will be set to low at the start of SPI messages and raised to high
#   after the message completes. This parameter must be provided.
#spi_bus: SPI3
#spi_speed:
spi_software_sclk_pin: PB3
spi_software_mosi_pin: PB5
spi_software_miso_pin: PB4

[tmc2130 extruder]
cs_pin: PD2
#   The pin corresponding to the TMC2130 chip select line. This pin
#   will be set to low at the start of SPI messages and raised to high
#   after the message completes. This parameter must be provided.
#spi_bus: SPI3
#spi_speed:
spi_software_sclk_pin: PB3
spi_software_mosi_pin: PB5
spi_software_miso_pin: PB4

My reference for these pins came from https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/blob/master/Hardware/SKR%20E3-DIP-PIN.pdf

r15ch13, 2019-10-15:
That worked! Thank you! 👍

closed 2019-10-19 by r15ch13