Klipper3d/klipper #2073: Fysetc cheetah 1.1  Printer is not ready

ice2009, 2019-10-14:
Hello :) 

I have probblem with klipper. I generate bin file for my board use stm32f103/no bootloader/no usb for communication and octoprint says Printer is not ready. I try generate bin with checked usb serial to communication and octoprit sasys same Printer is not ready . I upload bin file with stm32flash -w out/klipper.bin -v -i rts,-dtr,dtr /dev/ttyUSB0. In attachments i post log file and printer.cfg and .config file for generate bin. Could you help me please?

[klippy.log](https://github.com/KevinOConnor/klipper/files/3726198/klippy.log)

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/3726202/printer.cfg.txt)
[config for generate bin.txt](https://github.com/KevinOConnor/klipper/files/3726213/config.for.generate.bin.txt)

Thank's

BlackStump, 2019-10-15:
check this out below
https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

Then try change the flash command to 
> stm32flash -w out/klipper.bin -v -i rts,-dtr,dtr  /dev/serial/by-id/( change this to the port of your >board when you run the command ls -l /dev/serial/by-id/ )
> ie 
>stm32flash -w out/klipper.bin -v -i rts,-dtr,dtr  /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

ice2009, 2019-10-15:
Hello :) i try your command but octoprint say printer not ready 

here is flash command anyone help me ? :(

pi@octopi:~/klipper $ stm32flash -w out/klipper.bin -v -i rts,-dtr,dtr /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
stm32flash 0.5

http://stm32flash.sourceforge.net/

Using Parser : Raw BINARY
Interface serial_posix: 57600 8E1

GPIO sequence start
 setting port signal rts to 1... OK
 delay 100000 us
 setting port signal dtr to 0... OK
 delay 100000 us
 setting port signal dtr to 1... OK
GPIO sequence end

Version      : 0x22
Option 1     : 0x00
Option 2     : 0x00
Device ID    : 0x0414 (STM32F10xxx High-density)
- RAM        : Up to 64KiB  (512b reserved by bootloader)
- Flash      : Up to 512KiB (size first sector: 2x2048)
- Option RAM : 16b
- System RAM : 2KiB
Write to memory
Erasing memory
Wrote and verified address 0x08004914 (100.00%) Done.

KevinOConnor, 2019-10-19:
The "printer not ready" message could be due to many different reasons.  Attach the latest klipper log after the successful flash.

-Kevin