Klipper3d/klipper #2074: Printer stuck on heating

Kotletowsky, 2019-10-15:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3730977/klippy.log)
Ender + MKS Gen L 1.0 + RaspberryPi
Hi, after hotend upgrade I have huge problem. I've changed hotend from stock ender to triangle e3dv6. Both are 24V. Turn on printer, and set heating to 200C. Suddenly, printer go to 340C, so I disconnect power and wait till it cool down. Now, after every powerup it's still heating. Just for test I've changed extruder heater to anonther channel, and I have the same situation. Heater runaway. The same with old ender heater. I have no idea what's happened, board is broken ? But why ?

nophead, 2019-10-15:
Perhaps a short to ground in the heater cable somehow? Did you change the wiring?

Kotletowsky, 2019-10-15:
Yes, but for sure there is no shortcut. Even l crimp new connectors. Also with old heater is the same situation.

nophead, 2019-10-15:
No possibility you mixed up the heater and thermistor wires?

Kotletowsky, 2019-10-15:
No way. I'm electrician :)

nophead, 2019-10-15:
Does it all behave with the heater negative disconnected? I.e. stop heating and show correct temperature?

Seems like the MOSFET is shorted, they can be blown by ESD, but it would be odd to blow two.

Kotletowsky, 2019-10-15:
It's still showing correct temperature, but doesn't stop. Still heating.

nophead, 2019-10-15:
What even with one wire disconnected?

jakep82, 2019-10-15:
The MKS Gen L has 2 hotend outputs.  Move your hotend to the other output and change the pins in your config.  If it works correctly after that, then you know the mosfet on your board is shorted closed.

Kotletowsky, 2019-10-16:
If I disconnect one wire, mosfet is still triggered. About second heater output - the same situation. I've tried that. Also, when temperature goes higher than 230 firmware at printer freeze.

nophead, 2019-10-16:
If the board has a reset button hold that and see if the MOSFET turns off.

Did you change the firmware? If not how would it suddenly go wrong when you mess with the wiring? Looks more like you did something wrong with the wiring and blew up two MOSFETs.

Kotletowsky, 2019-10-16:
When I reset board mosfets are turned off, but start after release. About wiring. There are only 4 wires. It couldn't be wrong. Firmware reflashed yesterday.

nophead, 2019-10-16:
It now seems like a firmware heater pin configuration problem but you haven't made it clear what order you did things. If it failed simply by swapping extruders I can't see how it could be firmware. If you re-re-flashed the firmware before testing then it seems like the new firmware is miss-configured.

Kotletowsky, 2019-10-16:
First, just changed hotend. And it fails. After that, I thought it was a problem with firmware. So I reflashed that and set the same configuration like before. Also, I don't get why it freeze when temperature goes higher than 240. Thermal runaway doesn't work?

Kotletowsky, 2019-10-16:
Yes. I've set board fan to that output and extruder to second one (I have only one extruder). But now, board is dead. I need to wait for new one.

Kotletowsky, 2019-10-18:
Ok. New board, mks gen l 1.4. I've connected only raspberry, New heater, New thermistor and power supply. The same situation. Heater runaway, I can't stop this.

nophead, 2019-10-18:
You seem to have the extruder heater mapped to the fan connector, not the second extruder. Is your extruder heater wired to the fan connector? If not that looks like your problem.

Kotletowsky, 2019-10-19:
> 
> 
> You seem to have the extruder heater mapped to the fan connector, not the second extruder. Is your extruder heater wired to the fan connector? If not that looks like your problem.

I just use one extruder heater channel to drive fan for stepsticks. 
Now, I reinstaled octopi, octoprint, klipper host. Flashed new board from zero. I have only power supply connected, no stepsticks in sockets. And even i can't connect to printer: "ADC out of range"

nophead, 2019-10-19:
Yes but is the extruder heater that you say is always on actually connected to the fan connector on the board? That is where it is configured in your log of the first post. If you have the extruder heater wired to one of the heater connections it will not work with that configuration.

Kotletowsky, 2019-10-19:
Now I have everything complete fresh. And I can't even connect to printer. When I click "connect" immediately have "adc out of range".

nophead, 2019-10-19:
Well if you hadn't already said your are an electrician I would guess there is a problem with your thermistor connection, or it is configured for the wrong pin.

Kotletowsky, 2019-10-20:
New raspberry, New SD cards. Clean install and everything works as it should

closed 2019-10-20 by Kotletowsky