Klipper3d/klipper #2076: "Endstop and stepper must be on the same mcu," but they are

theophile, 2019-10-16:
I'm using an SKR v1.3 as the main controller and an SKR Mini v1.1 as a secondary controller to drive four additional steppers. After testing the system successfully and then cleaning up routing of wires, I decided that the Z endstop wire should be connected to the secondary mcu (shorter trip). That produced the "Endstop and stepper must be on the same mcu" error, so I swapped the Z stepper with one of the extra extruder steppers so that both the Z-axis stepper and endstop are connected to the same mcu. But it's still giving me the same "Endstop and stepper must be on the same mcu" error. Do _all_ steppers that have an endstop, and all their endstops, have to be connected to the same mcu?

https://pastebin.com/LY4XBuht

BlackStump, 2019-10-16:
probe I think needs to be on the same mcu as your Z stepper

theophile, 2019-10-18:
That was it. Thanks! My brain wasn't thinking about the bed probe as a Z-axis endstop although it obviously functions that way. It might be helpful if the error message would identify the pin that it choked on.

cxUnreal, 2021-04-23:
Referring to this issues ,how can I change the probe pin from mcu Z to mcu  in order to convenient for wiring and it won't giving me a error ?