Klipper3d/klipper #2077: Cetus - TinyFab CPU - Smoothie Pull

dharvey242, 2019-10-16:
Hi all, go easy on me.  Newbie here.  Can anyone shed any light on whether or not Klipper supports the Cetus TinyFab CPU.  From what I understand a bespoke version of smoothieware firmware has been used to support the TinyFab CPU and Cetus hardware.

https://github.com/tinyfab/Smoothieware

In particular the support for the PT100 i'm interested in.  I've used the generic smoothieware Make option in klipper and it doesn't appear to play nicely with the TinyFab CPU or Cetus Hardware or both.

I feel like the Klipper addition for the cetus would make a great addition give the limitation of the processor and the unlocked potential in the hardware.

dharvey242, 2019-10-16:
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/3733741/klippy.log.zip)
log file attached.

jakep82, 2019-10-16:
Assuming you have what's at the link, it appears to use the LPC1768 MCU.  That is supported in Klipper, but I have no idea if it's possible to flash that board, and you would most likely need to make your own config file.

https://www.tinyfab.xyz/product-page/tinyfab-up-cetus-mk3-cpu-usb

dharvey242, 2019-10-16:
Thanks for taking the time to help. Indeed it will flash to the board and I can communicate with it.  Started on my config file but the pt100 thermistor or thermocouple that is on the Cetus doesn't register the right temp.  It's shows about 20 degrees too low.  I believe it has something to do with the amplification of the signal that the tinyfab smoothie variation takes care of but I don't believe vanilla smoothie from klipper takes account of.

jakep82, 2019-10-16:
You may need to define a custom sensor.

https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L805

KevinOConnor, 2019-10-16:
Can you provide the working smoothieboard config file?

-Kevin

dharvey242, 2019-10-16:
# This file contains common pin mappings for Smoothieboard. To use
# this config, the firmware should be compiled for the LPC176x.

# See the example.cfg file for a description of available parameters.

[stepper_x]
step_pin: P2.1
dir_pin: P0.11
enable_pin: !P2.4
step_distance: .0125
endstop_pin: ^P1.24
#endstop_pin: ^P1.25
position_endstop: 0
position_max: 175
homing_speed: 50

[stepper_y]
step_pin: P2.0
dir_pin: !P0.5
enable_pin: !P2.4
step_distance: .0125
endstop_pin: ^P1.26
#endstop_pin: ^P1.27
position_endstop: 0
position_max: 175
homing_speed: 50

[stepper_z]
step_pin: P2.2
dir_pin: !P0.20
enable_pin: !P2.4
step_distance: .0025
endstop_pin: probe:z_virtual_endstop
#endstop_pin: ^P1.29
position_endstop: 0
position_max: 175

[extruder]
step_pin: P2.3
dir_pin: P0.22
enable_pin: !P2.4
step_distance: .002
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: P2.7
sensor_type: PT100 INA826
adc_voltage: 3.3
smooth_time: 2.0
sensor_pin: P0.23
control: pid
pid_Kp: 22
pid_Ki: 1.047
pid_Kd: 115
min_temp: 0
max_temp: 250

[heater_bed]
heater_pin: P0.4
sensor_type: NTC 100K beta 3950
sensor_pin: P0.26
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 130

[fan]
pin: P2.5

[mcu]
serial: /dev/ttyACM0

[printer]
kinematics: cartesian
max_velocity: 500
max_accel: 4000
max_z_velocity: 5
max_z_accel: 300

[bltouch]
sensor_pin: P2.13
control_pin: !P1.18
x_offset: 39
y_offset: 11
z_offset: 0.9
pin_up_touch_mode_reports_triggered: false
samples: 2
sample_retract_dist: 3.0

[homing_override]
gcode:
    G90 ; Use absolute position mode
    G1 Z10 ; Move up 10mm
    G28 X Y
    G1 X60 Y60 F6000 ; Change the X and Y coordinates to the center of your print bed
    G28 Z
set_position_z: 0.0

KevinOConnor, 2019-10-16:
Do you have the working config on smoothieware - not the klipper config file?

-Kevin

dharvey242, 2019-10-16:
# Smoothieboard configuration file, see http://smoothieware.org/configuring-smoothie
# NOTE Lines must not exceed 132 characters, and '#' characters mean what follows is ignored
## Robot module configurations : general handling of movement G-codes and slicing into moves

# Basic motion configuration
default_feed_rate                            4000             # Default speed (mm/minute) for G1/G2/G3 moves
default_seek_rate                            4000             # Default speed (mm/minute) for G0 moves
mm_per_arc_segment                           0.0              # Fixed length for line segments that divide arcs, 0 to disable
#mm_per_line_segment                         5                # Cut lines into segments this size
mm_max_arc_error                             0.01             # The maximum error for line segments that divide arcs 0 to disable
                                                              # note it is invalid for both the above be 0
                                                              # if both are used, will use largest segment length based on radius

# Arm solution configuration : Cartesian robot. Translates mm positions into stepper positions
alpha_steps_per_mm                           160               # Steps per mm for alpha stepper
beta_steps_per_mm                            160               # Steps per mm for beta stepper
gamma_steps_per_mm                           160               # Steps per mm for gamma stepper

# Planner module configuration : Look-ahead and acceleration configuration
# See http://smoothieware.org/motion-control
acceleration                                 3000             # Acceleration in mm/second/second.
#z_acceleration                              500              # Acceleration for Z only moves in mm/s^2, 0 uses acceleration which is the default. DO NOT SET ON A DELTA
junction_deviation                           0.05             # See http://smoothieware.org/motion-control#junction-deviation
#z_junction_deviation                        0.0              # For Z only moves, -1 uses junction_deviation, zero disables junction_deviation on z moves DO NOT SET ON A DELTA

# Cartesian axis speed limits
x_axis_max_speed                             30000            # Maximum speed in mm/min
y_axis_max_speed                             30000            # Maximum speed in mm/min
z_axis_max_speed                             300              # Maximum speed in mm/min

# Stepper module configuration 
# Pins are defined as  ports, and pin numbers, appending "!" to the number will invert a pin
# See http://smoothieware.org/pin-configuration and http://smoothieware.org/pinout
alpha_step_pin 2.1 #                         2.0              # Pin for alpha stepper step signal
alpha_dir_pin 0.11 #                         0.5              # Pin for alpha stepper direction
alpha_en_pin                                 2.4!             #0.4 # Pin for alpha enable pin
alpha_current                                1.5              # X stepper motor current
alpha_max_rate                               30000.0          # mm/min

beta_step_pin 2.0 #                          2.1              # Pin for beta stepper step signal
beta_dir_pin 0.5! #                          0.11             # Pin for beta stepper direction
beta_en_pin                                  2.4!             #0.10  # Pin for beta enable
beta_current                                 1.5              # Y stepper motor current
beta_max_rate                                30000.0          # mm/min

gamma_step_pin                               2.2              # Pin for gamma stepper step signal
gamma_dir_pin 0.20! #                        0.20!            # Pin for gamma stepper direction
gamma_en_pin                                 2.4!             # 0.19 # Pin for gamma enable
gamma_current                                1.5              # Z stepper motor current
gamma_max_rate                               300.0            # mm/min

## Extruder module configuration
# See http://smoothieware.org/extruder
extruder.hotend.enable                          true             # Whether to activate the extruder module at all. All configuration is ignored if false
extruder.hotend.steps_per_mm                    230              # up 640*1.53 Steps per mm for extruder stepper
extruder.hotend.default_feed_rate               600              # Default rate ( mm/minute ) for moves where only the extruder moves
extruder.hotend.acceleration                    500              # Acceleration for the stepper motor mm/sec²
extruder.hotend.max_speed                       50               # mm/s

extruder.hotend.step_pin                        2.3              # Pin for extruder step signal
extruder.hotend.dir_pin                         0.22             # Pin for extruder dir signal
extruder.hotend.en_pin                          2.4!             # 0.21 # Pin for extruder enable signal

# extruder offset
#extruder.hotend.x_offset                        0               # x offset from origin in mm
#extruder.hotend.y_offset                        0               # y offset from origin in mm
#extruder.hotend.z_offset                        0               # z offset from origin in mm

# firmware retract settings when using G10/G11, these are the defaults if not defined, must be defined for each extruder if not using the defaults
[... cut ...]
kill_button_pin                              2.12             # kill button pin. default is same as pause button 2.12 (2.11 is another good choice)

#msd_disable                                 false            # disable the MSD (USB SDCARD) when set to true (needs special binary)
#dfu_enable                                  false            # for linux developers, set to true to enable DFU

# Only needed on a smoothieboard
# See http://smoothieware.org/currentcontrol
currentcontrol_module_enable                 false             # Control stepper motor current via the configuration file

KevinOConnor, 2019-10-16:
Your smoothieware config has `temperature_control.hotend.ampmod1_pin 1.20`.  There is no equivalent in Klipper, but you could try permanently enabling that pin by adding to the config `[output_pin rtd_enable] pin: P1.20`.  However, I do not know if doing that will have adverse impacts.

-Kevin

KevinOConnor, 2019-10-16:
Actually, to attempt it, it would be:
```
[output pin rtd_enable]
pin: P1.20
value: 1
```

-Kevin

dharvey242, 2019-10-17:
I placed the three lines above the extruder section and tried again. Terminal came back with an error: Recv: !! Section 'output pin rtd_enable' is not a valid config section.  The temps also read zero on both the bed and hotend.  I suspect i'm doing something wrong in the config.

dharvey242, 2019-10-17:
To note the TinyFab cpu uses a fork of smoothie to work with the cetus hardware - https://github.com/tinyfab/Smoothieware.  Perhaps that can be the issue?

FHeilmann, 2019-10-17:
there's a typo in @KevinOConnors block. You need to add an underscore `_` so it becomes `output_pin` instead of `output pin`

dharvey242, 2019-10-17:
Changed it and put in the '_'.  Reverts back to where I was originally.  temp is reading 20 degrees less than it should.

KevinOConnor, 2019-10-19:
Can you attach the Klipper log file after adding the output_pin config section?

-Kevin

dharvey242, 2019-10-21:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3749222/klippy.log)

KevinOConnor, 2019-10-21:
Odd.  Despite the references to the "e3d pt100", the amplifier on that board doesn't match the e3d docs ( https://e3d-online.dozuki.com/Wiki/E3D_PT100_Amplifier_Documentation#Section_Output_Characteristics ).

So, it looks like you'll need to add a custom adc_temperature section - try something like:
```
[adc_temperature cetus_pt100_amp]
temperature1: -18.54
voltage1: 0.000
temperature2: 81.46 # 100 - 18.54
voltage2: 2.57604875
```

Then set the heater's sensor type to "cetus_pt100_amp".  See config/example-extras.cfg for further details.

-Kevin

KevinOConnor, 2019-10-21:
On further thought, the sample I gave before is unlikely to be correct.  The key data seems to be these two lines from the smoothieware config:
```
temperature_control.hotend.slope               0.0257604875
temperature_control.hotend.yintercept         -18.54
```
Unfortunately, I don't know what units these parameters are in, so it's difficult to translate them to the temperature/voltage (Celsius/Volts) that Klipper uses.

-Kevin

dharvey242, 2019-10-21:
The previous answer you gave me provided a -18 degrees temp.  So it did change something. Could I measure something coming in and out of the CPU?

dharvey242, 2019-10-21:
Further I found this from the author of the Cetus smoothie ware fork:

so, my change is T = a* X + b
where
X of course is ADC read
a = 0.0257604875
b = -18.54

KevinOConnor, 2019-10-21:
Maybe @arhi could help?  There's a desire to use Klipper with the Cetus CPU's PT100 amplifier.  However, it's not known how to convert the following smoothieware configuration:
```
temperature_control.hotend.ampmod1_pin         1.20           # UP! printer uses this to "energize" the RTD
temperature_control.hotend.ampmod2_pin nc #    1.21           # set as nc if you don't need to energize RTD
temperature_control.hotend.slope               0.0257604875
temperature_control.hotend.yintercept         -18.54
```

Klipper has a system (adc_temperature) for performing linear interpolation of temperatures, but the configuration is in Volts and Celsius.

Also, do the ampmod pins need to toggle at runtime, or can a static setting be used for them?

-Kevin

KevinOConnor, 2019-10-21:
This config may be a closer match:
```
[adc_temperature cetus_pt100_amp]
temperature1: -18.54
voltage1: 0.000
temperature2: 403.417   # 4095 * 4 * 0.0257604875 - 18.54
voltage2: 5.0
```

Do not specify an `adc_voltage` in the heater section.  Make sure to enable the stepper motors (eg, run `G28`) before inspecting the temperature.

If you test this, be careful as the reported temperatures may be totally inaccurate (the actual temperature may potentially be much higher than the reported temperature).

-Kevin

dharvey242, 2019-10-21:
okay temp is showing 8 degrees.  Bed temp is 24 which I think is correct.  This time will heat up and maintain a temp.  Whether it is accurately 180.  I'm not too sure how to tell.  Whenever i click extrude or retract it just shuts off all the power to the motors.   In fact G28 does the same.  All power is lost and it resets.  I can't see an error in the terminal.

arhi, 2019-10-21:
PT100 amp on the board is WEIRD .. I shared the schematic on few places .. 

- **ampmod pins**, I originally assumed these are used to not energize the PT100 sensor to not heat it from the measurement process, but it makes zero sense to have 2 pins especially on the schematic as they made it, so I believed that they are using ampmod pins for some type of calibration (read data with pin1, with pin1+2, with pin2, no pins, and then energize 2 pins to read the actual value) but I could not figure out wth was the process... then I seen bunch of other solutions they made and decided I just don't care and let's "hack it to work" so you see the result in https://github.com/arhi/Smoothieware

-**ampmod pins - always on**, no clue :( really, no idea, someone should try if keeping them always on would make any changes.. removing them from the mcu and hooking them directly to Vdd would release those 2 pins for something useful :D 

-** how to convert ...** well this 
```
temperature_control.hotend.slope               0.0257604875
temperature_control.hotend.yintercept         -18.54
```
is rather uber-simple line function :D ... the amp on the teartime board is pretty much linearizing the readout so I took 100 data points (what I read as raw data, what I input with PT100 emulator) put them in xls, drawn a line through them and from 190-250C error is below 5C .. and we are just talking about 5C readout error, not 5C drift, so maybe when you set it to 213C it's not really 213 but 215C in any way the PID will keep the temp constant and since difference of temp between tip of the nozzle and the temperature probe is over 20C according to thermal camera this was deemed, by me, acceptable :D or better "good enough" so don't read the temp from the box, print a temp tower and decide what temp readout on your printer is one that works best for that filament, who cares what "actual" temp that is :D 

so the output = read * slope + yintercept
Y = X * m + a, the line formula :) no thermistor curve no complications.... 

my firmware also support the actual PT100 RTD curve if you are reading data that is directly amplified and not linearized but that's wasting cpu cycles for no real reason on these machines.

the schematic (RA30 goes iirc trough ENABLE relay to Vdd too), so if someone can explain this resistor pattern to me, I'd be very grateful :D it makes no sense to me nor anyone I showed it to.
https://www.tiertime.com/community/viewtopic.php?f=28&t=22192&start=80#p47273

[image]

hopes this help, ping me if there's anything else I can help with

arhi, 2019-10-21:
btw, to use this T = ADC * 0.0257604875 -18.54  don't forget that 

```
ADC = readadc(heater0);
ADC += readadc(heater0);
ADC += readadc(heater0);
ADC += readadc(heater0);
T = ADC * 0.0257604875 -18.54
```

smoothieware is reading ADC 4 times and adding the values together (or reading 40 times then dividing by 10 or they do flowing average but anyhow you get your 10bit ADC * 4 value, not single 10bit read)

dharvey242, 2019-10-22:
okay temp reading is still about 6 degrees but if I manually enter temperature2: 650 is gets me to around the same temp as the bed at rest.  

Stepper motors however do nothing when I try to do G28.  They are energized but when i send a move command (G28) the device resets.  STEPPER_BUZZ STEPPER=stepper_y etc does the same thing.  Leads me to believe something is required as an additional stage or configuration for the steppers.  From what I have read they don't use I2C so at a lose as to what could be wrong.  Tmp file from what I can see doesn't provide any details on the failure other than failure to home.  I've checked the endstops and they are working fine.

KevinOConnor, 2019-10-23:
@dharvey242 - if you're having failures with STEPPER_BUZZ I would track that down first.  If you need assistance with that, attach the full klipper log file containing the event.

-Kevin

KevinOConnor, 2019-10-23:
@arhi - Thanks.  So, it sounds like an adc value of zero would result in a reported temperature of -18.54 and at a maximum adc value (4095 * 4) the reported temperature would be 403.417 (4095 * 4 * 0.0257604875 - 18.54).  That didn't result in the correct values for @dharvey242 , but it seems there might also be something else not working in that setup that should be tracked down first.

-Kevin

arhi, 2019-10-23:
yup, that's how it works on my UP plus 2 board ... not sure if all boards are same, I have up mini too and I noticed the board is different but were not checking out what type of amp it has but looking at some smoothieware configs tinyfab provided for mini pinout for the "cpu board" is different between mini and plus, possibly pt100 amp is different too... my suggestion is to modify the firmware to display raw adc values instead of temperature in C and just go trough the PT100 table, detach PT100 probe and if you don't have a PT100 emulator just attach a multiturn potentiometer. 200R would be ideal, 1k will do just good enough :D 
 - detach pot from the board
 - measure resistance of the pot, turn it to 138R5 (this is 100C) 
 - attach pot to board and read the ADC value
 - mark the XLS sheet with 100C, what you read as ADC value
 
do the same for 142R29 - 110C, 146R06 - 120C ... 175R84 - 200C ... 190R45 - 240C
(e.g. table from https://www.intech.co.nz/products/temperature/typert.html )

now plot the ADC vs TEMP curve and find a suitable line function that will best approximate segment between 190 and 250C (or whatever is the span of the temps one is working with on that printer) and use that for angle coeff. and y cross

KevinOConnor, 2019-11-08:
@dharvey242 - FYI, I added a new QUERY_ADC command to Klipper.  With that command, you should be able to read the raw analog values that Klipper is reading and then use that information to build the appropriate temperature table.  (For example, the command would be `QUERY_ADC NAME=extruder`.)

-Kevin

KevinOConnor, 2019-11-22:
I'm going to close this ticket as it appears the conversation has concluded.  Arhi - thanks for your help!

-Kevin

closed 2019-11-22 by KevinOConnor

arhi, 2019-11-22:
>  Arhi - thanks for your help!

any time guy's, thank you for running this great project