Klipper3d/klipper #2080: Parsing fails on bed_mesh probe_count param

mriley90, 2019-10-17:
Setting up bed mesh and Klipper continually fails to initialize properly. The error on the Octoprint terminal is `Internal error during connect: not enough arguments for format string`. I narrowed it down to the probe_count parameter, but my config for bed_mesh is the same as the example configs. I've tried several values for the probe count, including single values and pairs.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3737065/klippy.log)

BlackStump, 2019-10-17:
I don't think probe_count: 2,2 is right 3,3 is a minimum
odd numbers make more sense  as it will probe the bed center

mriley90, 2019-10-17:
I thought I tried 3,3 with the same error being reported, but I can't recreate that now, so I must have been mistaken. Sorry!

PR #2093 (2019-10-23): bed_mesh: configuration error bug fix
A minor bug fix to correct the config errors raised in the parse_pair() helper.  Resolves #2080 and #2035.
closed 2019-10-17 by mriley90