Klipper3d/klipper #2081: BLTouch and SKR1.3 - Z increments slightly every probe

nzlrhyz, 2019-10-17:
Hey I recently brought a genuine BLTouch to replace my Triangle labs clone and am getting an ever increasing Z probe you can easily see it in the probe accuracy test below Each test was run one after another with no actions inbetween.
I got better results with my clone than the genuine!
[probeaccuracy.txt](https://github.com/KevinOConnor/klipper/files/3737367/probeaccuracy.txt)

[klippy.log](https://github.com/KevinOConnor/klipper/files/3737364/klippy.log)
 -->

BlackStump, 2019-10-17:
10mm/s for probing is a tad quick try 5mm/s 
on a second note is your BLtouch mounted as per antclabs recommendations?

nzlrhyz, 2019-10-17:
I have also done it at 5mm/s with the same result. I can redo this if required 
I assume it is as I'm using the Reduplicator Titan Aero mount https://github.com/OmNomNomagon/ReDuplicator-MK3x
This all worked fine with the clone I literally swap out the clone with the genuine

BlackStump, 2019-10-17:
I have a older genuine classic bltouch and a trianglelabs clone of I think the antclabs bltouch-smart V2 and they both work fine,
I can only suggest make sure it is mounted correctly ie check the antclabs manual, make sure there is nothing loose and it doesn't have z flex seeing it offset quite a distance from the linear rods.
I am presuming it is a antclabs V3.1 the genuine?
what controller are you using?

nzlrhyz, 2019-10-17:
[probeaccuracy2.txt](https://github.com/KevinOConnor/klipper/files/3738320/probeaccuracy2.txt)
 I homed then ran two probe accuracy tests one after the other as you can see it increments even carrying on from the last probe accuracy then I turned motors off and rehomed which "reset" it back to what I believe is the true value 1.6.

Yes it is a 3.1
I'm not quite sure what you mean by controller

vladbabii, 2019-10-27:
@nzlrhyz that is not normal behaviour. You should get something along of 0.02 variation or less.

I would check your printer hardware based on the results, especially the Z axis couplers and the rods. what printer do you have ?

KevinOConnor, 2019-11-15:
This doesn't seem like a Klipper issue - it seems more like a hardware issue (eg, backlash or a loose grub screw).

You could try to verify by running GET_POSITION before and after each PROBE_ACCURACY run.

-Kevin

KevinOConnor, 2019-12-01:
I'm going to close this issue as it appears the conversation has concluded.

-Kevin

nzlrhyz, 2019-12-02:
Sorry I didn't receive any alerts for replies until yesterday, following your advice @KevinOConnor  heres the result
[probeaccuracy with get positions.txt](https://github.com/KevinOConnor/klipper/files/3913507/probeaccuracy.with.get.positions.txt)

KevinOConnor, 2019-12-05:
That log seems to show that the software did what it was intended to do.  Specifically, the mcu Z position stays constant (165) when the position is at Z=10.  So, klipper took the same number of steps each time, but the probe returned different results.

There's been reports of the bltouch probe needing to be adjusted (as described in docs/BL-Touch.md).  Otherwise, I don't have any further advice.

-Kevin

nzlrhyz, 2019-12-05:
I've just flashed marlin 2.0 on to the same machine and seems to be doing that same thing there too :S

KevinOConnor, 2019-12-13:
Okay, thanks.  I'm going to close this then as it looks like it is hardware related.

-Kevin

closed 2019-12-13 by KevinOConnor