Klipper3d/klipper #2083: Klipper loose control over heaters while homing

RustyWraith, 2019-10-18:
Performing G28 may lead to a large overheat or loss of extruder temperature. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/3745379/klippy.log)
[image]

KevinOConnor, 2019-10-19:
This is generally the result of voltage or wiring issues - for example: the power supply may not have sufficient power for the steppers and heaters (setting a max_power on the heaters may help), there could be poor ground wire connections, the axis movement may be tugging on the thermistor wire causing changes in voltage, the axis movement may be tugging on wires causing a general short, etc.

-Kevin

RustyWraith, 2019-10-19:
Thank you!
After some investigation I believe it was sort of thermistor issues.

closed 2019-10-19 by RustyWraith