Klipper3d/klipper #2084: Raspberry pi / ubuntu ???

Jerem1313, 2019-10-18:
hello, I use klipper for 8 months, very nice job, but I have a question.I installed a raspberry version of klipper with virtualbox (octopi) .Es this version will have the same performance as the version ubuntu? knowing that I use the emulator on pc. thank you in advance
sorry i'm french.

Jerem1313, 2019-10-19:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3746792/klippy.log)

KevinOConnor, 2019-10-19:
I don't know of anyone that has tested this, so you'll need to let us know how it impacts performance.  

-Kevin

vladbabii, 2019-10-27:
@Jerem1313 

I run klipper in
* ubuntu vmware virtual machine (windows host)
* ubuntu virtual box virtual machine (windows and linux host)
* docker on linux
* pi 3
* pi 4
* directly on ubuntu on a laptop

Klippers works fine on all as long as you
* do not hit a cpu limit while printing
* do not hit ram limit white printing
* can mount the serial connection inside docker/virtual machine

Basic rule is as long as the virtual/real machine has enough processing power and memory klipper will work fine.

If you have issues with a specific situation please give more details.

KevinOConnor, 2019-11-15:
I'm going to close this as it looks like conversation has concluded.

-Kevin

closed 2019-11-15 by KevinOConnor