Klipper3d/klipper #2086: Tmc5160 spi issue

Akex2, 2019-10-19:
hello
I have 2 tmc2208 for X and Y with uart and a tmc 5160 v1.2 from bitreetech in spi on a SKR1.3
[image]

The config with init command seems not work properly, when i do some init with dump, i have MRES = 14 or 13 ...
[klippy.log](https://github.com/KevinOConnor/klipper/files/3747107/klippy.log)
[image]
the cs pin is on extruder pin because i want try other place with the driver, for test it
[printer.cfg.log](https://github.com/KevinOConnor/klipper/files/3747109/printer.cfg.log)

KevinOConnor, 2019-10-19:
How did you connect the jumpers on the skr v1.3?  If there are any other connected devices that use spi then all must be configured in klipper.  I've also seen reports of various tmc5160 step stick boards that need modification to be used properly - I don't know if that is needed for "tmc 5160 v1.2 from bigreetech".

-Kevin

Akex2, 2019-10-20:
[image]

Akex2, 2019-10-20:
at the top is my board, and this is the setup for spi, i don't use sensorless
[image]

there aren't other spi device on that board

For the V1.2 of this driver, no need modification for use in spi.

i don,t know where is the error :(

vumario, 2019-10-20:
SKR board has SLP and RST pins connected together, but on TMC5160 those are CLK and SDO pins. To get SPI working correctly, you must cut off or otherwise disconnect CLK pin on your TMC5160 drivers

lenne0815, 2019-10-30:
> 
> 
> SKR board has SLP and RST pins connected together, but on TMC5160 those are CLK and SDO pins. To get SPI working correctly, you must cut off or otherwise disconnect CLK pin on your TMC5160 drivers

This is the right answer.

KevinOConnor, 2019-11-06:
Any further updates on this?  It sounds like the issue is thought to be a wiring issue with the stepsticks.

-Kevin

gaggi, 2019-11-11:
> there aren't other spi device on that board
> 
> For the V1.2 of this driver, no need modification for use in spi.
> 
> i don,t know where is the error :(

That is a misjudgment, they added a pull-up resistor on v1.2 that pulls the CLK pin low so you don't need to solder a jumper wire from clock to ground, but you still need to remove the clock pin.

Nandox7, 2019-11-11:
I also have the SKR v1.3 with BigTreeTechTMC5160 v1.2.

@gaggi so to confirm this is the pin I need to remove in the drivers?
[image]

gaggi, 2019-11-11:
@Nandox7 yes that is the pin that needs to be removed.

Akex2, 2019-11-12:
i already bended the pin, thanks at all for your help.
i think my driver has some hardware issue.

closed 2019-11-12 by Akex2