Klipper3d/klipper #2087: issue configuring display with skr 1.3

saurabhceaseless, 2019-10-20:
Hello

Im trying to interface the display to my skr 1.3 board.

I have a 12864 lcd display with uc1701 driver. There seems to be a configuration issue after restarting the klipper service. I receive an error with one of the EXP pin definitions saying error: Unknown value 'bt:P1.21' in enumeration 'pin'

I'm attaching my configuration file and klippy log file for reference.

I suppose P1.21 is a reserved pin in skr 1.3 board. I have additionally tried specifying the spi_bus under the display setting but that leads to an error stating spi_bus is invalid under display config.

The display type is a reprap discount type, I can hear the buzzer beeping when I click the encoder button, but display isn't working. This seems to be causing the issue is all I know for now.

Thanks 
Saurabh
[klippy.log](https://github.com/KevinOConnor/klipper/files/3747385/klippy.log)
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/3747386/printer.cfg.txt)

Arksine, 2019-10-20:
I haven't used pin aliases myself, but from looking at the log it appears that the aliases in the [board_pins] section should not contain the mcu prefix.  Instead, the prefix should be applied directly to the alias, for example:
```
[display]
lcd_type = uc1701
cs_pin = bt:EXP1_6
a0_pin = bt:EXP1_7
contrast = 40
encoder_pins = ^bt:EXP2_3, ^bt:EXP2_5
click_pin = ^!bt:EXP1_2
```

KevinOConnor, 2019-10-23:
Right.  If the goal is to define aliases for the "bt" micro-controller then you would define those aliases in a config section named `[board_pins bt]`.  There should not be a "bt:" prefix on the pins in the aliases list; there should be a "bt:" prefix in the pin names of the display config section.

-Kevin

saurabhceaseless, 2019-10-24:
Hello

Tried the above, still getting the same error, please refer log attached -

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/3767749/printer.cfg.txt)
[klippy.log](https://github.com/KevinOConnor/klipper/files/3767750/klippy.log)

Thanks
Saurabh

KevinOConnor, 2019-11-06:
I'm going to close this as it looks like the answer to the original question was answered.

-Kevin

closed 2019-11-06 by KevinOConnor

saurabhceaseless, 2019-11-09:
Hello

Im still trying to get the generic fysetc display to work with skr 1.3 board.

I have verified the pin mapings on the board and the display, Im just trying to assign the right pins from the board to the display assigning aliases to pins or not.

I re-tried by removing aliases to pins and directly stating pins but I keep getting error that a particular pin is reserved for ssp0 -

Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 138, in _connect
    cb()
  File "/home/pi/klipper/klippy/mcu.py", line 672, in _connect
    self._check_config()
  File "/home/pi/klipper/klippy/mcu.py", line 613, in _check_config
    self._send_config(None)
  File "/home/pi/klipper/klippy/mcu.py", line 576, in _send_config
    self._config_cmds[i] = pin_resolver.update_command(cmd)
  File "/home/pi/klipper/klippy/pins.py", line 195, in update_command
    return re_pin.sub(pin_fixup, cmd)
  File "/home/pi/klipper/klippy/pins.py", line 193, in pin_fixup
    name, self.reserved[pin_id]))
error: pin P0.17 is reserved for ssp0

How can I remove this reservation?

Thanks
Saurabh
[klippy.log](https://github.com/KevinOConnor/klipper/files/3826838/klippy.log)

KevinOConnor, 2019-11-09:
If you have a display similar to "MKS Mini 12864 LCD" (see config/sample-lcd.cfg) then the config should look like:
```
[board_pins bt]
aliases:
    # EXP1 header
    EXP1_1=P1.30, EXP1_3=P1.18, EXP1_5=P1.20, EXP1_7=P1.22, EXP1_9=<GND>,
    EXP1_2=P0.28, EXP1_4=P1.19, EXP1_6=P1.21, EXP1_8=P1.23, EXP1_10=<5V>,
    # EXP2 header
    EXP2_1=P0.17, EXP2_3=P3.26, EXP2_5=P3.25, EXP2_7=P1.31, EXP2_9=<GND>,
    EXP2_2=P0.15, EXP2_4=P0.16, EXP2_6=P0.18, EXP2_8=<RST>, EXP2_10=<NC>
    # Pins EXP2_1, EXP2_6, EXP2_2 are also MISO, MOSI, SCK of bus "ssp0"

[display]
lcd_type: uc1701
cs_pin: bt:EXP1_6
a0_pin: bt:EXP1_7
contrast: 40
encoder_pins: ^bt:EXP2_3, ^bt:EXP2_5
click_pin: ^!bt:EXP1_2

[output_pin beeper]
pin: bt:EXP1_1
```

Alternatively, if you have the "Fysetc Mini 12864Panel v2.1" display, it should look like:
```
[board_pins bt]
aliases:
    # EXP1 header
    EXP1_1=P1.30, EXP1_3=P1.18, EXP1_5=P1.20, EXP1_7=P1.22, EXP1_9=<GND>,
    EXP1_2=P0.28, EXP1_4=P1.19, EXP1_6=P1.21, EXP1_8=P1.23, EXP1_10=<5V>,
    # EXP2 header
    EXP2_1=P0.17, EXP2_3=P3.26, EXP2_5=P3.25, EXP2_7=P1.31, EXP2_9=<GND>,
    EXP2_2=P0.15, EXP2_4=P0.16, EXP2_6=P0.18, EXP2_8=<RST>, EXP2_10=<NC>
    # Pins EXP2_1, EXP2_6, EXP2_2 are also MISO, MOSI, SCK of bus "ssp0"

[display]
lcd_type: st7567
cs_pin: bt:EXP1_3
a0_pin: bt:EXP1_4
rs_pin: bt:EXP1_5
contrast: 63
encoder_pins: ^bt:EXP2_3, ^bt:EXP2_5
click_pin: ^!bt:EXP1_2

[output_pin beeper]
pin: bt:EXP1_1

[neopixel fysetc_mini12864]
pin: bt:EXP1_6
chain_count: 3
color_order_GRB: False
initial_RED: 0.4
initial_GREEN: 0.4
initial_BLUE: 0.4
```

If you have some other type of display, then you'll need to tell us what the pins are for that display.

-Kevin

saurabhceaseless, 2019-11-09:
Hello Kevin

Im using a fysetc generic display, the pinouts are defined below by the manufacturer -

https://wiki.fysetc.com/Generic_12864_Panel/
[image]

Now trying to map that to skr 1.3 pinouts for EXP1 and 2, there is one caveat, the EXP1 and 2 has to be reversed by connecting the connector other way around as described by another issue.
[image]

Comparing the above two, I get this pin mappings (correct me if Im wrong) -
cs_pin = P1.18
a0_pin = P1.19
encoder_pins = P3.26, P3.25
click_pin = P0.28
beeper pin = P0.17

The manufacturer says that the display is a ST7567 type but also compatible as uc1701 type -
"This screen driver is ST7567 and is compatible with the marlin Mini panel (UC1701,same as ST7567). "

I'll try configuring as ST7567 and try again as per your input.

Thanks
Saurabh