Klipper3d/klipper #2088: Hello

Kenzu, 2019-10-20:
Hello
I have a sculpto printer. It has custom electronics, based on an linkit smart7688 wifi module with openwrt and an LPC1768 (smoothie based). They talk via uart.
The machine is not normal Polar, but more like bipolar (like an old LP player). I have a smoothie and marlin port working on the machine, but would like to try out klipper.
How to get started?

[https://sculpto.dk](url)

[image]

_Originally posted by @Kenzu in https://github.com/KevinOConnor/klipper/pull/1023#issuecomment-544189457_

KevinOConnor, 2019-10-20:
This would require a developer to implement the kinematics.  There's some info on writing new kinematics at https://www.klipper3d.org/Code_Overview.html#adding-new-kinematics .

Do you know the math formulas for the "inverse kinematics" and "forward kinematics"?

-Kevin

Kenzu, 2019-10-22:
I short like this...
Will update later. :-)

cartesian_to_actuator
theta2 = 2 * asinf( sqrtf(x*x+y*y) / (2*arm_length) );
theta1 = (PI-theta2)/2 - atan2f(y,x);

actuator_to_cartesian
theta = ((PI-theta2_rad)/2) - theta1_rad;
 r = 2 * arm_length * sinf(theta2_rad/2);
//Convert from polar to cartesian
x = r * cosf(theta);
y = r * sinf(theta);

KevinOConnor, 2019-10-23:
Seems simple.  A "bipolar" implementation would look similar to commit ec9cb3a1, but with a different implementation for polar_stepper_radius_calc_position(), polar_stepper_angle_calc_position(), and calc_position().

Let us know if you need assistance.

-Kevin

KevinOConnor, 2019-11-24:
FYI, I looked a little closer at this, and the step generation math is basically the same as a scara robot.  Which is also very similar to the recent work on the rotary delta kinematics (the work-rotary-20191119 branch).

-Kevin

KevinOConnor, 2019-12-13:
It doesn't look like this ticket is under active development, so I'm going to mark it as inactive.

-Kevin

closed 2019-12-13 by KevinOConnor