Klipper3d/klipper #2089: Printer variables for temperature_fan not working

B0rax, 2019-10-20:
I am trying to display the temperature of my chamber thermistor on my LCD screen while heating up the printer. For this I created a g-code macro:
`[delayed_gcode chamber_display]
gcode:
        M117 Chamber: { printer["temperature_fan chamber"].temperature }`

(temperature_fan is configured and works correctly)

Now, when I want to run the macro, nothing happens. Is there something I am doing wrong? Klippy.log gives me the following error:

> Error evaluating 'delayed_gcode chamber_display:gcode': UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'temperature_fan chamber'
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 61, in render
    return str(self.template.render(context))
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/jinja2/environment.py", line 1008, in render
    return self.environment.handle_exception(exc_info, True)
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/jinja2/environment.py", line 780, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "<template>", line 2, in top-level template code
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/jinja2/environment.py", line 430, in getattr
    return getattr(obj, attribute)
UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'temperature_fan chamber'
Script running error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/extras/delayed_gcode.py", line 34, in _gcode_timer_event
    self.gcode.run_script(self.timer_gcode.render())
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 66, in render
    raise self.gcode.error(msg)
CommandError: Error evaluating 'delayed_gcode chamber_display:gcode': UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'temperature_fan chamber'

Arksine, 2019-10-21:
Currently the temperature_fan module does not report its status, thus it has no attributes that can be accessed using templates in the config file.  It is a simple fix that I have resolved in PR #2095.

B0rax, 2019-10-21:
Thank you very much for that quick fix!

PR #2095 (2019-10-23): temperature_fan: add get_status() function
Add status reporting for template usage.  Resolves #2089.
closed 2019-10-23 by KevinOConnor